VOLTTRON/volttron #1005: Scope out a Zigbee driver

jhaack, 2016-10-28:
Investigate what it would take to implement a Zigbee driver, possibly using https://github.com/flyte/xbee-helper.

kmonson, 2016-11-01:
Unlike BACnet or Modbus there is no standard for ZigBee over IP or a standard API for interfacing with a ZigBee device. This means that supporting ZigBee will mean supporting one or more specific ZigBee devices. 

Most likely the first device we will support is the Xbee. Xbee works over serial and uses a simple async protocol to communicate with the device. This will need to be wrapped in a ZigBee Proxy. The python driver is found here: https://github.com/nioinnovation/python-xbee

Basic ZigBee support will require the following:

Spec out a generic way to represent points on a ZigBee device and a way to address the device. This will require understanding the fundamentals of how zigbee devices are represented so that any proxy will be able to work with a standard address representation.

Spec out an generic RPC interface for a ZigBee Proxy.

Create the generic ZigBee driver.

Create a Xbee based ZigBee proxy.

Device auto discovery features exist and would be a bonus.

schandrika, 2020-02-26:
Related to #1016 Something that can be done if the community is interested?