VOLTTRON/volttron #1017: local vip pubsub publish hangs if platform not running.

carlatpnl, 2016-11-02:
the local volttron platform must be running else these calls will hang:
        self.vip.pubsub.publish('pubsub', msgtopic, header, msg )
                                               or
        self.vip.pubsub.publish('pubsub', msgtopic, header, msg ).get(timeout=2)

This can be inconvenient when doing local debugging which does not require the platform.

craig8, 2016-11-03:
@carlatpnl  Can you describe when you say local debugging?  Because not matter what when I am using an agent I have to connect to a platform instance in order to use it's services.  Pubsub being one of those services.

I am curious if this may be related to #1018 in which there is a running through of an event.wait(timeout=5) rather than wrapping an if condition.

carlatpnl, 2016-11-03:
by 'local debugging' I am referring to using Eclipse to step through code in an agent or driver after hitting a set breakpoint.  In this particular case i was testing the formatting of a data structure my driver was sending to the 'C' daemon in controls.  When I continue from my breakpoint, other parts of the driver eventually call the pubsub.publish mentioned earlier.