VOLTTRON/volttron #1018: Forwarder's historian setup does not raise gevent.Timeout

mikeroup, 2016-11-02:
The `event.wait(timeout=10)` will block for up to ten seconds and return false if the event is not set.

mikeroup, 2016-11-02:
Other instances of this pattern:
- [ ] `examples/Alerter/send.py`
- [ ] `volttrontesting/utils/platformwrapper.py`
- [ ] `volttron/platform/vip/agent/core.py` (used correctly)
- [ ] `volttron/platform/vip/agent/connection.py`
- [ ] `volttron/platform/vip/agent/utils.py`
- [ ] `services/core/ForwardHistorian/forwarder/agent.py`
- [ ] `services/core/MasterDriverAgent/master_driver/interfaces/universal.py`

craig8, 2016-11-03:
@mikeroup do you mind if I make the changes to some of these?  I think most of them actually were my code!

craig8, 2016-11-03:
@mikeroup  See  #1020 for coments on how we need to revamp the ForwardHistorian.  However for the 4.0.1 release lets just do the minimal and use the utility functions for creating agents or connections to the external platform.

closed 2017-01-03 by craig8