VOLTTRON/volttron #1041: MasterDriverAgent not using the values of campus/building/unit, only the key names.

ocschwar, 2016-11-11:
I have a master driver agent set to scrape a ModbusTCP slave. The modbus config file is as follows: 

`{
    "driver_config": {"device_address": "127.0.0.1", "port":5020},
    "campus": "Shirley-MA",
    "building": "North",
    "unit": "PMC",
    "driver_type": "modbus",
    "registry_config":"/home/volttron/Docs/design-docs/voltron master modbus agent registers.csv",
    "interval": 5,
    "timezone": "UTC",
    "heart_beat_point": "ESMMode"
}
`

Yet the topics by which the driver posts things are just "campus/building/modbus1/SolMetering_L1_to_L2_AC_Voltage" 

I changed volttron/services/core/MasterDriverAgent/master_driver/driver.py as follows to try to fix it, but to no avail:

`        self.base_topic = DEVICES_VALUE(campus=config.get('campus',''),
                                        building=config.get('building',''),
                                        unit=config.get('unit',''),
                                        path=self.device_path,
                                        point=None)
        
        self.device_name = DEVICES_PATH(campus=config.get('campus',''),
                                        building=config.get('building',''),
                                        unit=config.get('unit',''),
                                        base='',
                                        node='',
`

craig8, 2016-11-12:
this is not using the config store...from what I can tell.  What branch are you using?

ocschwar, 2016-11-12:
The develop branch.

ocschwar, 2016-11-12:
I mistakenly thought that removing a wheel and re-installing it would nuke the configuration store.

ocschwar, 2016-11-12:
Converting to the configuration store fixed the issue. Thanks!

closed 2016-11-12 by ocschwar

craig8, 2016-11-12:
Great!