VOLTTRON/volttron #1048: Add additional functionality to basehistorian

craig8, 2016-11-14:
As an optional argument to the basehistorian there should be two additional arguments: filter and transform.

The transform argument would contain a list of tuples e.g.

````
[
    ("from1", "to1"),
    ("orange", "grape")
]
````

In this scenario the content from1 and orange would be replaced in any topic that contained it to to1 and grape respectively.

The filter argument would contain a list strings that must be in the topic in order to be stored. 

````
["foo", "bin"]
````

For the topic devices/AMCE/BUILDING1/foo would be stored but the topic devices/ACME/BUILDING1/baf would not be included.

craig8, 2016-11-14:
Open for thoughts.

schandrika, 2020-02-26:
We do topic string replace already. Is the filter option necessary? We can add custom topic list to control which topics the historian agent records.

jhaack, 2020-02-26:
I think we could close this. If someone expresses a current need for this functionality we can create a new ticket.

closed 2020-02-26 by schandrika