VOLTTRON/volttron #1053: Review and possibly remove some global arguments from volttron-ctl

zbeech, 2016-11-16:
`volttron-ctl` is a bit cluttered with global arguments. We should review these arguments to see if they are necessary or helpful. Candidates for removal include:

- `--config`
- `--timeout`
- `--debug`
- `--developer-mode`
- `--log-config`
- `--quite`
- `--verbose`
- `--verboseness`

zbeech, 2016-11-16:
`--vip-address` is necessary for controlling a remote platform

mikeroup, 2016-11-16:
`--developer-mode` is only useful if the main Volttron process is started with it also.  For the next release we could probably remove `--developer-mode` entirely.

mikeroup, 2016-11-16:
The volttron-ctl log is separate from the main volttron log and is pretty sparse. The `--verbose` option can be useful but the `--log`, `--log-config`, `--quiet`, and `--verboseness` options seem like overkill.

craig8, 2016-11-16:
As an aside I believe --developer-mode will be going away, not sure if it is going to be going away in 4.1 or 4.2 as yet.

schandrika, 2020-02-26:
@shwethanidd could you verify this when you work on vctl auth update fopr csr