VOLTTRON/volttron #1060: RPC function calls not getting encoded properly.

ocschwar, 2016-11-22:
I have an agent with one RPC.export'ed method: 
```

  @RPC.export("twiddle")
    def set_setpoint(self, setpoint, *args, **kwargs):
        self.setpoint = setpoint
        _log.info("Twiddled to %s"%self.setpoint)

```

When I use VolttronCentral to invoke , or when I try to call inspect on it: 

`{
    "method": "platforms.uuid.f6e675fb36989f97c3b0f25227aaf02e.agents.cffe568d-7ebc-4b67-8d6c-9d8d49c28d8e.inspect",
    "params": {
        "setpoint": 1
    },
    "jsonrpc": "2.0",
    "id": "36f0bc60-b069-11e6-ac3f-c181716e6964",
    "authorization": "1c21e5bf-8501-4985-8533-9f268368d687"
}
` 
I get a timeout, and this in volition.log: 

> 2016-11-22 04:07:42,187 (volttroncentralagent-3.5.5 18578) volttroncentral.agent INFO: rpc method: platforms.uuid.f6e675fb36989f97c3b0f25227aaf02e.agents.cffe568d-7ebc-4b67-8d6c-9d8d49c28d8e.inspect
2016-11-22 04:07:42,211 (vcplatformagent-3.5.4 18457) <stdout> INFO: ('RPC ERROR', 'Traceback (most recent call last):\n  File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 168, in method\n    return method(*args, **kwargs)\n  File "/home/volttron/.volttron/agents/e6621e64-58c7-40d9-afe9-c2bc695220a4/vcplatformagent-3.5.4/vcplatform/agent.py", line 732, in route_request\n    result = self.vip.rpc.call(identity, agent_method, **params).get(timeout=30)\n  File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 306, in call\n    self.core().socket.send_vip(peer, \'RPC\', [request], msg_id=ident)\n  File "/home/volttron/volttron/volttron/platform/vip/socket.py", line 428, in send_vip\n    flags=flags|more, copy=copy, track=track)\n  File "/home/volttron/volttron/volttron/platform/vip/socket.py", line 402, in send_multipart\n    msg_parts, flags=flags, copy=copy, track=track)\n  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/zmq/green/core.py", line 250, in send_multipart\n    msg = super(_Socket, self).send_multipart(*args, **kwargs)\n  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/zmq/sugar/socket.py", line 363, in send_multipart\n    i, rmsg,\nTypeError: Frame 0 (None) does not support the buffer interface.\n')
2016-11-22 04:07:42,212 (volttroncentralagent-3.5.5 18578) <stderr> ERROR: Traceback (most recent call last):
2016-11-22 04:07:42,212 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:   File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/gevent/greenlet.py", line 534, in run
2016-11-22 04:07:42,212 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:     result = self._run(*self.args, **self.kwargs)
2016-11-22 04:07:42,212 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:   File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 258, in _handle_subsystem
2016-11-22 04:07:42,213 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:     dispatch(bytes(msg), message) for msg in message.args) if response]
2016-11-22 04:07:42,213 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:   File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 258, in <genexpr>
2016-11-22 04:07:42,213 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:     dispatch(bytes(msg), message) for msg in message.args) if response]
2016-11-22 04:07:42,214 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:   File "/home/volttron/volttron/volttron/platform/jsonrpc.py", line 354, in dispatch
2016-11-22 04:07:42,214 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:     return self.serialize(response)
2016-11-22 04:07:42,214 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:   File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 95, in serialize
2016-11-22 04:07:42,214 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:     return jsonapi.dumps(json_obj)
2016-11-22 04:07:42,215 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:   File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/zmq/utils/jsonapi.py", line 40, in dumps
2016-11-22 04:07:42,215 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:     s = jsonmod.dumps(o, **kwargs)
2016-11-22 04:07:42,215 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:   File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/simplejson/__init__.py", line 397, in dumps
2016-11-22 04:07:42,215 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:     **kw).encode(obj)
2016-11-22 04:07:42,216 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:   File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/simplejson/encoder.py", line 275, in encode
2016-11-22 04:07:42,216 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:     chunks = self.iterencode(o, _one_shot=True)
2016-11-22 04:07:42,216 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:   File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/simplejson/encoder.py", line 357, in iterencode
2016-11-22 04:07:42,216 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:     return _iterencode(o, 0)
2016-11-22 04:07:42,217 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:   File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/simplejson/encoder.py", line 252, in default
2016-11-22 04:07:42,217 (volttroncentralagent-3.5.5 18578) <stderr> ERROR:     raise TypeError(repr(o) + " is not JSON serializable")
2016-11-22 04:07:42,217 (volttroncentralagent-3.5.5 18578) <stderr> ERROR: TypeError: TypeError('Frame 0 (None) does not support the buffer interface.') is not JSON serializable
2016-11-22 04:07:42,218 (volttroncentralagent-3.5.5 18578) <stderr> ERROR: <Greenlet at 0x7f30990d39b0: _handle_subsystem(<volttron.platform.vip.agent.subsystems.rpc.RPC ob, Message(**{'peer': 'master.web', 'subsystem': 'RPC)> failed with TypeError
2016-11-22 04:07:42,218 (volttroncentralagent-3.5.5 18578) <stderr> ERROR: 
2016-11-22 04:07:42,206 (vcplatformagent-3.5.4 18457) volttron.platform.vip.agent.subsystems.rpc ERROR: unhandled exception in JSON-RPC method 'route_request': 
Traceback (most recent call last):
  File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 168, in method
    return method(*args, **kwargs)
  File "/home/volttron/.volttron/agents/e6621e64-58c7-40d9-afe9-c2bc695220a4/vcplatformagent-3.5.4/vcplatform/agent.py", line 732, in route_request
    result = self.vip.rpc.call(identity, agent_method, **params).get(timeout=30)
  File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 306, in call
    self.core().socket.send_vip(peer, 'RPC', [request], msg_id=ident)
  File "/home/volttron/volttron/volttron/platform/vip/socket.py", line 428, in send_vip
    flags=flags|more, copy=copy, track=track)
  File "/home/volttron/volttron/volttron/platform/vip/socket.py", line 402, in send_multipart
    msg_parts, flags=flags, copy=copy, track=track)
  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/zmq/green/core.py", line 250, in send_multipart
    msg = super(_Socket, self).send_multipart(*args, **kwargs)
  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/zmq/sugar/socket.py", line 363, in send_multipart
    i, rmsg,
TypeError: Frame 0 (None) does not support the buffer interface.

craig8, 2016-11-22:
Please remove twiddle from the export.  

````
@RPC.export
def set_setpoint(self, setpoint, *args, **kwargs):
````

or

````
@RPC.export("set_setpoint")
def set_setpoint(self, setpoint, *args, **kwargs):
````

or

````
@RPC.export(set_setpoint)
def set_setpoint(self, setpoint, *args, **kwargs):
````

The code mentioned with "twiddle" is expecting a function called twiddle to exist in the class.

ocschwar, 2016-11-22:
Same error nonetheless.

And it also happens when I try to call the "inspect" method instead.

ocschwar, 2016-11-22:
Just tried this: 

`{
"method": "platforms.uuid.f6e675fb36989f97c3b0f25227aaf02e.agents.uuid.f4947d9e-6a81-4d62-90c7-db5e08d4b0f1.set_setpoint",
"params": {
    "setpoint":1
    },
    "jsonrpc": "2.0",
    "id": "41f3d010-b03d-11e6-bc13-73f06deffd5e",

     "authorization": "2394842f-`
[...]

And still got this: 

`2016-11-22 15:38:26,447 (vcplatformagent-3.5.4 22405) vcplatform.agent DEBUG: Calling method set_setpoint on agent f4947d9e-6a81-4d62-90c7-d
b5e08d4b0f1
2016-11-22 15:38:26,447 (vcplatformagent-3.5.4 22405) vcplatform.agent DEBUG: Params is: {'setpoint': 1}
2016-11-22 15:38:26,444 (volttroncentralagent-3.5.5 22493) volttroncentral.agent INFO: rpc method: platforms.uuid.f6e675fb36989f97c3b0f25227
aaf02e.agents.uuid.f4947d9e-6a81-4d62-90c7-db5e08d4b0f1.set_setpoint
2016-11-22 15:38:26,444 (volttroncentralagent-3.5.5 22493) volttroncentral.agent DEBUG: REMOTE_ADDR: 209.104.240.250
2016-11-22 15:38:26,444 (volttroncentralagent-3.5.5 22493) volttroncentral.agent DEBUG: SESSION_USER IS: {'ip': '209.104.240.250', 'token': 
'2394842f-8adb-4c84-bab8-85819a28ee8c', 'user': 'admin', 'groups': 'admin'}
2016-11-22 15:38:26,444 (volttroncentralagent-3.5.5 22493) volttroncentral.agent DEBUG: RPC METHOD IS: platforms.uuid.f6e675fb36989f97c3b0f2
5227aaf02e.agents.uuid.f4947d9e-6a81-4d62-90c7-db5e08d4b0f1.set_setpoint
2016-11-22 15:38:26,445 (volttroncentralagent-3.5.5 22493) volttroncentral.agent DEBUG: inside _route_request b5d5edb0-b0c9-11e6-8cb3-cb15eb
2d3b80, platforms.uuid.f6e675fb36989f97c3b0f25227aaf02e.agents.uuid.f4947d9e-6a81-4d62-90c7-db5e08d4b0f1.set_setpoint, {'setpoint': 1}
2016-11-22 15:38:26,445 (volttroncentralagent-3.5.5 22493) volttroncentral.agent DEBUG: f6e675fb36989f97c3b0f25227aaf02e
2016-11-22 15:38:26,445 (volttroncentralagent-3.5.5 22493) volttroncentral.agent DEBUG: Routing to platform.agent
2016-11-22 15:38:26,445 (volttroncentralagent-3.5.5 22493) volttroncentral.agent DEBUG: Routing request b5d5edb0-b0c9-11e6-8cb3-cb15eb2d3b80 agents.uuid.f4947d9e-6a81-4d62-90c7-db5e08d4b0f1.set_setpoint {'setpoint': 1}
2016-11-22 15:38:26,482 (vcplatformagent-3.5.4 22405) <stdout> INFO: ('RPC ERROR', 'Traceback (most recent call last):\n  File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 168, in method\n    return method(*args, **kwargs)\n  File "/home/volttron/.volttron/agents/e6621e64-58c7-40d9-afe9-c2bc695220a4/vcplatformagent-3.5.4/vcplatform/agent.py", line 732, in route_request\n    result = s:
elf.vip.rpc.call(identity, agent_method, **params).get(timeout=30)\n  File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 306, in call\n    self.core().socket.send_vip(peer, \'RPC\', [request], msg_id=ident)\n  File "/home/volttron/volttron/volttron/platform/vip/socket.py", line 428, in send_vip\n    flags=flags|more, copy=copy, track=track)\n  File "/home/volttron/volttron/volttron/platform/vip/socket.py", line 402, in send_multipart\n    msg_parts, flags=flags, copy=copy, track=track)\n  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/zmq/green/core.py", line 250, in send_multipart\n    msg = super(_Socket, self).send_multipart(*args, **kwargs)\n  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/zmq/sugar/socket.py", line 363, in send_multipart\n    i, rmsg,\nTypeError: Frame 0 (None) does not support the buffer interface.\n')
2016-11-22 15:38:26,476 (vcplatformagent-3.5.4 22405) volttron.platform.vip.agent.subsystems.rpc ERROR: unhandled exception in JSON-RPC method 'route_request': 
Traceback (most recent call last):
  File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 168, in method
    return method(*args, **kwargs)
  File "/home/volttron/.volttron/agents/e6621e64-58c7-40d9-afe9-c2bc695220a4/vcplatformagent-3.5.4/vcplatform/agent.py", line 732, in route_request
    result = self.vip.rpc.call(identity, agent_method, **params).get(timeout=30)
  File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 306, in call
    self.core().socket.send_vip(peer, 'RPC', [request], msg_id=ident)
  File "/home/volttron/volttron/volttron/platform/vip/socket.py", line 428, in send_vip
    flags=flags|more, copy=copy, track=track)
  File "/home/volttron/volttron/volttron/platform/vip/socket.py", line 402, in send_multipart
    msg_parts, flags=flags, copy=copy, track=track)
  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/zmq/green/core.py", line 250, in send_multipart
    msg = super(_Socket, self).send_multipart(*args, **kwargs)
  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/zmq/sugar/socket.py", line 363, in send_multipart
    i, rmsg,
TypeError: Frame 0 (None) does not support the buffer interface.

`

ocschwar, 2016-11-22:
Added some logging to show that the identity in that vip.rpc.call is set to None, which ZMQ appears not to like.

ocschwar, 2016-11-22:
So, I was persistently getting the UUID wrong. Once the agent UUID is correct, the problem goes away.

craig8, 2016-11-22:
Ahh yes.  This is an issue that is addressed much better in the 4.0.1 code to be released soon.  The platform uuid is deterministic and the agent uuid is coupled very closely with the identity.

ocschwar, 2016-11-22:
Ok, thank you.

closed 2016-11-22 by ocschwar