VOLTTRON/volttron #1062: Add api method to get vendor from vendor ID

jchap-pnnl, 2016-11-22:
Here's the current vendor table. It would be good for the method to return this same object:

{

    "0": "ASHRAE",

    "1": "NIST",

    "2": "The Trane Company",

    "3": "McQuay International",

    "4": "PolarSoft",

    "5": "Johnson Controls, Inc.",

    "6": "American Auto-Matrix",

    "7": "Siemens Schweiz AG (Formerly: Landis & Staefa Division Europe)",

    "8": "Delta Controls",

    "9": "Siemens Schweiz AG",

    "10": "Schneider Electric",

    "11": "TAC",

    "12": "Orion Analysis Corporation",

    "13": "Teletrol Systems Inc.",

    "14": "Cimetrics Technology",

    "15": "Cornell University",

    "16": "United Technologies Carrier",

    "17": "Honeywell Inc.",

    "18": "Alerton / Honeywell",

    "19": "TAC AB",

    "20": "Hewlett-Packard Company",

    "21": "Dorsette’s Inc.",

    "22": "Siemens Schweiz AG (Formerly: Cerberus AG)",

    "23": "York Controls Group",

    "24": "Automated Logic Corporation",

    "25": "CSI Control Systems International",

    "26": "Phoenix Controls Corporation",

    "27": "Innovex Technologies, Inc.",

    "28": "KMC Controls, Inc.",

    "29": "Xn Technologies, Inc.",

    "30": "Hyundai Information Technology Co., Ltd.",

    "31": "Tokimec Inc.",

    "32": "Simplex",

[... 1778 lines cut ...]
    "922": "Tyco Fire & Security S.p.A.",

    "923": "Watanabe Electric Industry Co., Ltd.",

    "924": "Causam Energy",

    "925": "W-tec AG",

    "926": "IMI Hydronic Engineering International SA",

    "927": "ARIGO Software",

    "928": "MSA Safety",

    "929": "Smart Solucoes Ltda - MERCATO",

    "930": "PIATRA Engineering",

    "931": "ODIN Automation Systems, LLC",

    "932": "Belparts NV",

    "999": "Reserved for ASHRAE"

}

craig8, 2017-01-03:
This basically moves the vendors to the backend and makes them available rather than having hardcoded them into the javascript code.

Theodore100billy, 2023-04-26:
How do I get the information I need from ODIN Automation Systems, LLC concerning mygov logins