VOLTTRON/volttron #1066: BACpypes configuration file BBMD and FD

loulupe, 2016-11-24:
I'm trying to scan devices from a laptop to a BACnet BBMD structure with several Bacnet routers/IP.
All the points are network visible, and I have the BACnet routers IP address.

I have got to scan the devices under the router using Yabe using this configuration:
Local addresss like 192.168.1.2
Foreign device address 10.10.10.7
UDP 47808

But I cannot find the right configuration for BACpypes.
I downloaded the updated BACpypes and go through the BBMD2VLANRouter.py (as I found on a question from your team to BACpypes about BBMD configuration https://sourceforge.net/p/bacpypes/mailman/message/34587436/)

I have tried several number configurations for this file, but I have no luck with getting a BVLC message answer in wireshark.

Could you elaborate what is the right configuration in the BACini file for cases with BBMD or FD??
Do you have an example for this?

craig8, 2016-11-24:
@loulupe this question is more suited for Stackoverflow than here on the issues.  Can you post this there? @kmonson will most surely be able to answer you when he gets back to work on Monday.

loulupe, 2016-11-24:
@craig8 Hi Craig and happy thanksgiving!
I choose to put it here, because I think this should be addressed as a configuration file issue for VOLTTRON for integration with existing building automation system with more complex architecture that include BACnet/IP Broadcast Management Device (BBMD).  

The last similar request from Kyle to BACPypes required an update of BACPypes. 

This is a spreedsheet that shows different types of architecture, it would be very useful to have documentation to know which ones are covered with BACPypes, and how the configuration should be set up.

https://docs.google.com/spreadsheets/d/1neArwaeN1Ro4Q8kHVqnRYa_9P4QqCGTbsZ7FXKwiNEg/edit#gid=0

kmonson, 2016-11-28:
Hi @loulupe!

Can I get the following from you:

The BACpypes.ini file you are using.
The IP address of your notebook computer on the network.
The IP of the target device.
The IP of the BBMD.

If the subnet masks are anything other than 255.255.255.0 I'll want to know that too.

loulupe, 2016-11-29:
@kmonson 
Thanks Kyle, I cloned the BACpypes from git repository https://github.com/JoelBender/bacpypes.git
The architecture is a laptop connected to the network to a BBMD router to a BACnet/Arcnet with a target device eg. VAV controller, with similar addresses to this.

The PC on the network is 192.168.1.2/24:47808
The BBMD is 10.7.10.1/24:47808  (BACnet/IP router)
The Bacnet/Arcnet Address 1/Network 7100
The target device Address 40 / Network 7101

Subnet mask is 24 -  255.255.255.0

rlutes, 2016-11-29:
@loulupe 
Is the pc that is performing the scan on a subnet that is configured within the BBMD?  If you do the scan directed at a specific IP address (for a BACnet/IP device) does the scan "see" the device?

loulupe, 2016-11-30:
@rlutes @kmonson 
The PC/laptop is on the public campus network and the Bacnet routers (LGR) are on the VLAN.
I have been able to scan the router and devices using Yabe, whit a configuration of Foreign Device Address for the BBMD router. 
Is there a similar configuration for VOLTTRON?  for Foreign devices / BBMD?

rlutes, 2016-12-09:
If you register the computer as a foreign device on the BBMD the IP address should then be routable to the BBMD.  The VOLTTRON instance should be able to communicate with devices behind the router.  Is the VOLTTRON instance using the same IP and BACnet device instance registered by the YABE tool?  Does the YABE tool unregister on connection close?

rlutes, 2016-12-16:
Were you able to come to resolution on this?   If so, please comment with the resolution so others may benefit from the knowledge.