VOLTTRON/volttron #1079: File too long error with ZMQ IPC

jhaack, 2016-12-02:
It seems like the IPC changes to include auth can cause filenames to be too large. This causes an error when trying to use volttron-ctl status and other commands through ipc.

Traceback (most recent call last):
  File "/home/aaosensor/git/volttron/env/local/lib/python2.7/site-packages/gevent/greenlet.py", line 534, in run
    result = self._run(*self.args, **self.kwargs)
  File "/home/aaosensor/git/volttron/volttron/platform/vip/agent/core.py", line 286, in run
    loop = looper.next()
  File "/home/aaosensor/git/volttron/volttron/platform/vip/agent/core.py", line 636, in loop
    self.socket.connect(self.address)
  File "/home/aaosensor/git/volttron/volttron/platform/vip/socket.py", line 541, in connect
    addr.connect(self, super(_Socket, self).connect)
  File "/home/aaosensor/git/volttron/volttron/platform/vip/socket.py", line 258, in connect
    (connect_fn or sock.connect)(self.base)
  File "zmq/backend/cython/socket.pyx", line 528, in zmq.backend.cython.socket.Socket.connect (zmq/backend/cython/socket.c:5644)
  File "zmq/backend/cython/checkrc.pxd", line 25, in zmq.backend.cython.checkrc._check_rc (zmq/backend/cython/socket.c:9542)
    raise ZMQError(errno)
ZMQError: File name too long
<Greenlet at 0xa5b202c: <bound method Core.run of <volttron.platform.vip.agent.core.Core object at 0xa5b06ec>>(<gevent.event.Event object at 0xa5b45cc>)> failed with ZMQError

jhaack, 2016-12-02:
This has only been observed on Ubuntu 32 bit (11.10 and 12.04). Have not been able to replicate on other 32 bit installs.

zbeech, 2016-12-03:
I'm not yet sure how to solve this, but here's a little test that might be helpful:

```python
import zmq 

context = zmq.Context()

print("Server test:")
socket = context.socket(zmq.REP)
try:
    socket.bind("ipc://" + "A"*200)
except zmq.ZMQError as e:
    print(e)

print("--------------------------------------------")

print("Client test:")
socket = context.socket(zmq.REQ)
try:
    socket.connect("ipc://" + "B"*200)
except zmq.ZMQError as e:
    print(e)
```

mikeroup, 2016-12-03:
Strange that this happens on a connect after the platform has already started. Why doesn't it happen on bind?

zbeech, 2016-12-03:
It looks like the root of the problem is in python's `urlparse` module. (It is used in `volttron/platform/vip/socket.py` to parse the socket parameters.) Here is a simple test:

```python

import urlparse
address = "ipc://@/home/vip.socket?param1=A&param2=B"
url = urlparse.urlparse(address, 'tcp')
print(url)
```

My system is running python 2.7.12, on which the test outputs:

```
ParseResult(scheme='ipc', netloc='@', path='/home/vip.socket', params='', query='publickey=pub&secretkey=sec&serverkey=serv', fragment='')
```

A problematic system I tested is running python 2.7.3. On that system the test outputs:

```
ParseResult(scheme='ipc', netloc='@', path='/home/vip.socket?param1=A&param2=B', params='', query='', fragment='')
```
It does not correctly parse the query from the URL.

jhaack, 2016-12-03:
We'll have to patch it similar to what we had to do with the bad sqlite library in old Python versions.

closed 2016-12-08 by mikeroup