VOLTTRON/volttron #1108: Compiling the VC ui-src

ocschwar, 2017-01-03:
I'm trying to merge the latest commit to develop, to add my own stuff, and I see the gulp file is gone and web pack is in. 

What compilation steps are supposed to be done here now?

jchap-pnnl, 2017-01-03:
From the VolttronCentral directory, run **npm run build-watch** to run the compilation script. That script is defined in package.json along with a few other scripts that we left in for now but may still remove or alter.

If you use the **npm run build-watch** command and make changes to the ui-src code, it will hot reload the changes, but you'll need to refresh the browser to see them. A side effect is you'll get extra css.map and js.map files in webroot. You can clean those out by deleting them manually or canceling the script and rerunning it, which will wipe out webroot and rebuild it.

ocschwar, 2017-01-03:
Thank you for the information. I am at the moment getting this, however: 

```
/home/volttron/volttron/services/core/VolttronCentral/webpack.config.js:5
const ExtractTextPlugin = require('extract-text-webpack-plugin')
SyntaxError: Use of const in strict mode.

^^^^^

```

ocschwar, 2017-01-03:
And removing "use strict;" from web pack.config.js, after asking for a bunch of npm installs,
gives me this: 

`/home/volttron/volttron/services/core/VolttronCentral/node_modules/case-sensitive-paths-webpack-plugin/index.js:72
    var parsedPath = path.parse(dir);
                          ^
TypeError: Object #<Object> has no method 'parse'
    at CaseSensitivePathsPlugin.fileExistsWithCaseSync (/home/volttron/volttron/services/core/VolttronCentral/node_modules/case-sensitive-paths-webpack-plugin/index.js:72:27)
    at NormalModuleFactory.<anonymous> (/home/volttron/volttron/services/core/VolttronCentral/node_modules/case-sensitive-paths-webpack-plugin/index.js:125:34)
    at NormalModuleFactory.applyPluginsAsyncWaterfall (/home/volttron/volttron/services/core/VolttronCentral/node_modules/webpack/node_modules/tapable/lib/Tapable.js:86:13)
    at onDoneResolving (/home/volttron/volttron/services/core/VolttronCentral/node_modules/webpack/lib/NormalModuleFactory.js:38:11)
`

jchap-pnnl, 2017-01-03:
Not sure if you tried this already, but do the following:

1. delete your VolttronCentral/node_modules directory.
2. run the command **npm install --save** from the VolttronCentral directory to get all the dependencies and rebuild node_modules.
3. then run the command **npm run build-watch** to run the script. 

You shouldn't have to remove "use strict" from the webpack.config, so hopefully updating the dependencies will fix it. Let us know if not. Thanks.

jchap-pnnl, 2017-01-03:
Correction: don't use **--save** in step 2.

ocschwar, 2017-01-03:
That and upgrading Node.js to 7 fixed the build issue.
Thanks!

 Now I just have to figure out why VC is giving 404 for index.html

craig8, 2017-01-03:
I am going to assume that you have rebuilt the vc agent and have rebuilt, bootstrapped your volttron environment.  If not please run python bootstrap.py from your activated environment.

If you want to see changes directly from the npm run build-watch you can add an absolute path to the webroot in your VolttronCentral/volttroncentral/webroot directory into the config file for volttron central.

Let me know if this helps.

ocschwar, 2017-01-04:
Problem turned out that the platform agent also needed updating. Going to try bootstrapping now to address some remaining issues.

closed 2017-01-04 by ocschwar