VOLTTRON/volttron #1112: perhaps browser memory leak with charts

craig8, 2017-01-03:
After having charts running for a while I end up with 
[image]

I will  leave the developer window open to see what if anything is shown there.

craig8, 2017-01-04:
The picture with the Websockets was a configuration issue with our external vc host.  

However with the latest develop and releases/4.0.1 code I have the following 

[image]

jchap-pnnl, 2017-02-13:
Commits 27f0be2, 60024b0, and a347d05 pertained to this issue.

craig8, 2017-03-04:
Merged new chart lib in #1229

closed 2017-03-04 by craig8