VOLTTRON/volttron #1123: MasterDriverAgent not setting registers.

ocschwar, 2017-01-05:
I have a Modbus-TCP slave (written with pymodbus) as a single device, with a long list of registers. 

The  MDA is set to scrape this one device, and does so fine. 

But when I use the Actuator agent, I can reserve the slave correctly, call set_multiple_points, 
and get back the successful return value (an empty dict), but on the slave side, no points are actuated.

They are all set as writeable on both ends. 

This is the 'develop' branch.

ocschwar, 2017-01-05:
The code I'm using:

`    def reserve_modbus (self):

        start = datetime.now().strftime(
            "%Y-%m-%d %H:%M:%S")
        end = (datetime.now()+timedelta(minutes=2)).strftime(
            "%Y-%m-%d %H:%M:%S")
        return self.vip.rpc.call(
            "platform.actuator",
            "request_new_schedule",
            "", "ess"+self.PATH, "HIGH",
            [self.PATH, start, end]).get()

    def release_modbus(self):
        return self.vip.rpc.call(
            "platform.actuator",
            "request_cancel_schedule",
            "curtail","ess"+self.PATH).get()

    @RPC.export
    def set_site(self, target_min, target_max, *args, **kwargs):
        """                                                                                                                                                   
        Set a plant wide maximum power production level (real power)                                                                                          
"""
        res = self.reserve_modbus()
        _log.info("Reserved: {}".format(res))
        _log.info("DEMO Set SITE bounds to {}<->{}".format(target_min,target_max))
        values = [
            (self.PATH + "ESS.Control.RealControl.Mode",self.SITE_TARGET),
            (self.PATH + "ESS.Control.SiteControl.MinPower",int(target_min)),
            (self.PATH + "ESS.Control.SiteControl.MaxPower",int(target_max))
        ]
        _log.info("Values %s"%values)
        ret = self.vip.rpc.call(
            "platform.actuator",
            "set_multiple_points",
            "ess"+self.PATH,
            values)
`

ocschwar, 2017-01-05:
Is there a reason why e second IOError is not raised?

`    def _set_point(self, point_name, value):    
        register = self.get_register_by_name(point_name)
        if register.read_only:
            raise  IOError("Trying to write to a point configured read only: "+point_name)

        with modbus_client(self.ip_address, self.port) as client:
            try:
                result = register.set_state(client, value)
            except (ConnectionException, ModbusIOException, ModbusInterfaceException) as ex:
                IOError("Error encountered trying to write to point {}: {}".format(point_name, ex))
        return result
`

rlutes, 2017-01-05:
Can you post the registry csv (or part if it is long) for the Modbus slave?

ocschwar, 2017-01-06:
The registers I'm looking at are these: 

`  {
    "Point Address": "40809", 
    "Unit Details": "", 
    "Modbus Register": ">i", 
    "Writable": "TRUE", 
    "Volttron Point Name": "ESS.Control.RealControl.Mode", 
    "Default Value": "", 
    "Units": "", 
    "Variance": null, 
    "Notes": ""
  }, 
  {
    "Point Address": "40780", 
    "Unit Details": "", 
    "Modbus Register": ">i", 
    "Writable": "TRUE", 
    "Volttron Point Name": "ESS.Control.SiteControl.MaxPower", 
    "Default Value": "", 
    "Units": "W", 
    "Variance": null, 
    "Notes": "Only in \"Direct-Control\" Mode:  Pass through command per Tesla Spec"
  }, 
  {
    "Point Address": "40779", 
    "Unit Details": "", 
    "Modbus Register": ">i", 
    "Writable": "TRUE", 
    "Volttron Point Name": "ESS.Control.SiteControl.MinPower", 
    "Default Value": "", 
    "Units": "W", 
    "Variance": null, 
    "Notes": "Only in \"Direct-Control\" Mode:  Pass through command per Tesla Spec"
  }, 
`

And so far as my slave sees, the write-holding-register command comes with a value of 0 regardless of the values I send.

ocschwar, 2017-01-07:
Wrote this server to verify I wasn't crazy. 

`from pymodbus.server.async import StartTcpServer
from pymodbus.datastore import ModbusSequentialDataBlock,ModbusSparseDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
import argparse
import sys
import logging

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

class SlaveContext(ModbusSlaveContext):
    def setValues(self, fx, address, values):
        ret = super(SlaveContext,self).setValues(fx, address,values)
        chk = super(SlaveContext,self).getValues(fx, address,len(values))
        log.error("RET {} {} {} {} {}".format(self.decode(fx), ret,address, values, chk))
        return ret
    

def main(argv):
    # run the server you want
    #---------------------------------------------------------------------------#
    parser = argparse.ArgumentParser(
        description = "Start a PMC Simulator. Quit with SIGINT")
    opts,args = parser.parse_known_args(argv)
    port = int(args[1])
    block = ModbusSequentialDataBlock.create()
    store = SlaveContext(
        di = block,
        co=block,
        ir=block,
        hr=block)
    context = ModbusServerContext(slaves=store, single=True)
    StartTcpServer(context, address=("localhost", port))

    
if __name__ == "__main__":
    main(sys.argv)
`

ocschwar, 2017-01-07:
Ah. struct.Struct('>I') is asking for trouble. Sorry to bother you all.

closed 2017-01-07 by ocschwar

craig8, 2017-01-07:
:) glad you got it figured.  @kmonson was going to write you a response, but is out of the lab today.