VOLTTRON/volttron #1128: Enable SunSpec in the MasterDriverAgent. (32 bit registers and longer)

ocschwar, 2017-01-07:
SunSpec, along with many inverters, now use Modbus register allocations where a single item is sprawled across multiple registers. These include big-endian 32 and 64 bit integers, and string allocations for model numbers and serial numbers. 

For Modbus to set these registers correctly, modbus.py, needs to go from this:
`    
    def set_state(self, client, value):
        if not self.read_only:   
            value_bytes = self.parse_struct.pack(value)
            register_values = PYMODBUS_REGISTER_STRUCT.unpack_from(value_bytes)
            client.write_registers(self.address, register_values, unit=self.slave_id)
            return self.get_state(client)
        return None    
`
To something like this:

`    def set_state(self, client, value):
        if not self.read_only:   
            value_bytes = self.parse_struct.pack(value)
            register_values = []
            for i in range(0, len(value_bytes),2):
                register_values += PYMODBUS_REGISTER_STRUCT.unpack_from(
                    value_bytes[1:1+1])
            client.write_registers(self.address, register_values, unit=self.slave_id)
            return self.get_state(client)
        return None
`

Sunspace talks about writing to such registers, and it's convenient to be able to have the MDA do it for simulation purposes even when the 32 bit register in question is meant to be read-only. I've run into a couple holding registers that are listed as 32 bit.

ocschwar, 2017-01-07:
https://github.com/FraunhoferCSE/volttron/commit/bd0ffe150fab90d437e5ed56c07b8870f41c0bac

This commit is in on my fork, and it works. I am now using 32 bit writeable registers.

kmonson, 2017-01-12:
This is a bug. The  driver was always intended to support this case. I even tested writing 32 bit values against a real device. Looking at the code now it is obvious that is won't work properly in almost all cases. It appears that I was just unlucky with my test and missed this.

There are a few things in your solution that need to be cleaned up. I'll do that and get it into develop later today.

kmonson, 2017-01-12:
Would you try changing the code to this and testing it:

    def set_state(self, client, value):
        if not self.read_only:   
            value_bytes = self.parse_struct.pack(value)
            register_values = []
            for i in xrange(0, len(value_bytes), PYMODBUS_REGISTER_STRUCT.size):
                register_values.extend(PYMODBUS_REGISTER_STRUCT.unpack_from(value_bytes, i))
            client.write_registers(self.address, register_values, unit=self.slave_id)
            return self.get_state(client)
        return None

This should be cleaner. I don't have a device to test against right this second. I just need to know that it works and then I'll put in a pull request.

Your commented out code was the more correct solution. It just had a few typos and needed a few small tweaks.

ocschwar, 2017-01-13:
Just tested, and it works.

kmonson, 2017-01-13:
Fixed in 5f753994a1226f4833c2ef8b3354af42c0fce8f2

closed 2017-01-13 by kmonson