VOLTTRON/volttron #1137: VOLTTRON Central Chart Point Selection UI

acedrew, 2017-01-12:
If you're adding charts to a page that already has charts, and you're using the filter functionality to search the point list, the list will refresh at intervals, causing the entire list to be displayed again, and losing your scroll position.

jchap-pnnl, 2017-01-13:
Hi @acedrew, I'm not sure what you mean that the list will refresh at intervals causing it to be displayed again. When you say list, are you referring to the tree, and it's the tree that refreshes? Can you provide further explanation and a screen shot or two to help me understand the scenario and how to reproduce it?

jchap-pnnl, 2017-01-13:
Or do you mean the list in the Add Chart dialog box? If so, screen shots would still be helpful. Thanks.

ocschwar, 2017-01-13:
I can confirm encountering the same behavior. Click "add chart", and the dialog appears. The topics selector gets populated a couple seconds later. While you're typing something to narrow the list of topics, the topics list is refreshed again, and instead of looking at the narrowed down list, you're looking at the full one. 

Just click "add chart", start typing a topic string in, watch the list narrow down, and then stop. It will grow back.

jchap-pnnl, 2017-01-13:
OK, I think I'm understanding the problem now. Thanks. I'll investigate and see if I can get it fixed.