VOLTTRON/volttron #1142: Where is version 4.0 document?

so3500, 2017-01-13:
I am studying version 4.0 of Voltron. But the document is only 3.x. Name of agent (ex. PlatformAgent -> VolttronCentralPlatform). Where can I get the 4.0 version of the Voltron document?

craig8, 2017-01-13:
http://volttron.readthedocs.io/en/4.0.1/ all of the current documentation is up on the readthedocs site.  You can get any of the versions from there.  There is a link on the bottom left of the page that has the version of the document going with the release.

The next release will be 4.1 after a mini release of 4.0.1 which will be transitioned to master within the next week.

so3500, 2017-01-13:
Thank you for your reply.
[image]

However, when i click Downloads / PDF, the Release 3.5RC1 document appears.
Should I then only rely on the site?

craig8, 2017-01-13:
Once you go to one of the other versions I believe the PDF will be what you want.  Note the PDF is just a rendering of what is online

so3500, 2017-01-13:
Thank you very much for your quick reply.

closed 2017-01-13 by so3500