VOLTTRON/volttron #1144: MySQL Historian query syntax issue.

ocschwar, 2017-01-13:
I just migrated my Sql Historian to MySQL, and now when I try to extend a graph size in VoltTron Central, there is this: 

```

2017-01-13 19:59:22,445 (volttroncentralagent-3.6.0 32108) <stderr> ERROR: TypeError: mysql.connector.errors.ProgrammingError(1064, "1064 (42000): You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near ''2000'' at line 5", '42000') is not JSON serializable
2017-01-13 19:59:22,446 (volttroncentralagent-3.6.0 32108) <stderr> ERROR: <Greenlet at 0x7fdd3819dd70: _handle_subsystem(<volttron.platform.vip.agent.subsystems.rpc.RPC ob, Message(**{'peer': 'master.web', 'subsystem': 'RPC)> failed with TypeError
2017-01-13 19:59:22,446 (volttroncentralagent-3.6.0 32108) <stderr> ERROR: 
2017-01-13 19:59:22,449 (sqlhistorianagent-3.6.1 28216) <stdout> INFO: ('RPC ERROR', 'Traceback (most recent call last):\n  File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 168, in method\n    return method(*args, **kwargs)\n  File "/home/volttron/volttron/volttron/platform/agent/base_historian.py", line 1205, in query\n    agg_period, skip, count, order)\n  File "/home/volttron/.volttron/agents/46c22212-9910-4ec1-950b-780a434d2d42/sqlhistorianagent-3.6.1/sqlhistorian/historian.py", line 306, in query_historian\n    count=count, order=order)\n  File "/home/volttron/volttron/volttron/platform/dbutils/mysqlfuncts.py", line 290, in query\n    rows = self.select(real_query, args)\n  File "/home/volttron/volttron/volttron/platform/dbutils/basedb.py", line 458, in select\n    cursor.execute(query, args)\n  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/mysql/connector/cursor.py", line 515, in execute\n    self._handle_result(self._connection.cmd_query(stmt))\n  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/mysql/connector/connection.py", line 490, in cmd_query\n    result = self._handle_result(self._send_cmd(ServerCmd.QUERY, query))\n  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/mysql/connector/connection.py", line 395, in _handle_result\n    raise errors.get_exception(packet)\nProgrammingError: 1064 (42000): You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near \'\'2000\'\' at line 5\n')

```

craig8, 2017-01-13:
Can you provide your mysql version, and operating system all that good information would help us figure this out.

ocschwar, 2017-01-13:
Debian Jessie, MySQL 5.5

ocschwar, 2017-01-13:
2017-01-13 20:23:52,018 (sqlhistorianagent-3.6.1 28799) volttron.platform.dbutils.mysqlfuncts WARNING: Real Query: SELECT topic_id, ts, value_string
                FROM data
                WHERE topic_id = %s
                 ORDER BY topic_id DESC, ts DESC
                LIMIT %s
                
2017-01-13 20:23:52,018 (sqlhistorianagent-3.6.1 28799) volttron.platform.dbutils.mysqlfuncts WARNING: args: [66, '200']

So the issue is that the count parameter is never forced to be an int. 

mysqlfuncts.py, base_historian.py and sqlhistorian/historian.py all have places where the 
enforcement could be done.

craig8, 2017-01-13:
would you consider making a patch and committing that back?

ocschwar, 2017-01-13:
Sure thing. Policy wise, where should that go?

craig8, 2017-01-13:
You can submit a pull request against develop.  I think that's what you are asking.

ocschwar, 2017-01-13:
https://github.com/FraunhoferCSE/volttron/commit/3e781e8814ad67e6a93a63a9ae0416993b8419d9

Tested and works.

ocschwar, 2017-01-13:
I was asking where in the call stack the type enforcement should go, but since None is allowed for count, there's only one place where it can go.

closed 2017-01-19 by ocschwar