VOLTTRON/volttron #1150: Forward Historian logs that it has handled items, but items do not show up in target platform log

acedrew, 2017-01-14:
Forward historian is capturing data, and claims it is handled, but it never shows up in the target platform log. No errors are being generated.

craig8, 2017-01-14:
Platform log or historian?

acedrew, 2017-01-14:
The items show in the collector platform logs as being "handled" by the
forward historian, however on the central platform logs, they don't appear
at all, nor do the make it into the database

acedrew, 2017-01-14:
From collector:

Truncated.....
 u'Drivers.BacnetNetwork.LG DEVICE 6.points.ACL_40.JanES_ACL40_LIBa_MDE_ST': {u'units': u'State', u'type': u'integer', u'tz': u''}, u'Drivers.BacnetNetwork.LG DEVI
CE 6.points.ACL_40.JanES_ACL40_LIBa_ALM': {u'units': u'Enum', u'type': u'integer', u'tz': u''}, u'Drivers.BacnetNetwork.LG DEVICE 6.points.ACL_40.JanES_ACL40_LIBa_ZN
_T': {u'units': u'degreesFahrenheit', u'type': u'integer', u'tz': u''}, u'Drivers.BacnetNetwork.LG DEVICE 6.points.ACL_42.JanES_ACL42_LIBc_F_SPD_CMD': {u'units': u'S
tate', u'type': u'integer', u'tz': u''}, u'Drivers.BacnetNetwork.LG DEVICE 6.points.ACL_41.JanES_ACL41_LIBb_MDE_CMD': {u'units': u'State', u'type': u'integer', u'tz'
: u''}}]}
2017-01-14 07:29:05,675 (forwarderagent-3.7 5557) forwarder.agent DEBUG: handled: 8 number of items
2017-01-14 07:29:05,676 (forwarderagent-3.7 5557) volttron.platform.agent.base_historian DEBUG: Cleaning up successfully published values.
2017-01-14 07:29:05,694 (forwarderagent-3.7 5557) volttron.platform.agent.base_historian DEBUG: Getting oldest outstanding to publish.
2017-01-14 07:29:05,695 (forwarderagent-3.7 5557) volttron.platform.agent.base_historian DEBUG: Reading from/waiting for queue.

acedrew, 2017-01-19:
@craig8 can you reference the PR/Commit you fixed this in? Then we can close.
Working correctly in latest develop branch

craig8, 2017-01-19:
Referencing the fix for the base historian #1154

closed 2017-01-19 by craig8