VOLTTRON/volttron #1162: Register sub volttron instance to central volttron instance and how to message with each other instance

so3500, 2017-01-24:
http://volttron.readthedocs.io/en/4.0.1/devguides/walkthroughs/VOLTTRON-Central-Demo.html#remove-charts

I tried VOLTTRON Central Demo with the link above.

### And I tried to connect a remote VOLTTRON instance.
The figure I first thought was as shown below.
[image]

I refer to the volttron user guide3.0, volttron document.

#### When registering the sub voltron in the center volttron, the following information is input.
serverkey : public key of center volttron
public key, private key, vip address : sub volttron

[image]

And failed.
[image]

I do not know what went wrong. I do not know whether to refer to document and user guide. Please let me know.

### And I know that using ForwardAgent after connecting will communicate between volttrons. How do I send a message or data?

craig8, 2017-01-24:
The forwarderagent takes the publishes from the message bus under the analysis/, datalogger/, device/ or record/ topics and publishes them to a second instance.  The second instance MUST have been authorized in order to connect to the first instance.  See http://volttron.readthedocs.io/en/4.0.1/devguides/walkthroughs/VOLTTRON-Central-Demo.html#getting-started for the authentication of a remote connection.

craig8, 2017-01-24:
There are a couple things to watch the logs for if you see a failed authentication error in the log.  This will happen when an agent has the correct publickey to talk to the volttron instance, but doesn't have permissions to access the volttron instance.  You can add that publickey that is listed in the log or add the /.*/ to the auth file and then remove it after all of the volttron central platforms (vcp)s have been registered.

sonjinho, 2017-01-25:
Thank you we solved this problem

we didn't open the port about 22916 & 8080 

And didn't need key(public ,server, secret) for register just auto register platform

craig8, 2017-01-25:
Great ports open would definitely be an issue for connecting

closed 2017-01-25 by craig8