VOLTTRON/volttron #1165: Add support for querying by source

kmonson, 2017-01-30:
Currently it is not possible to specify the source for data when doing a query. 

For example if a device published to devices/foo/bar and an Agent published record/foo/bar. If you query for foo/bar you will get results from both sources. 

This means that we will also need to add source support to the concrete historians.

craig8, 2017-01-31:
This seems to be a problem with the concrete historian implementation.  The only one that has a collision is analysis and devices.  The rest of the topics can use the root node aka (record/, datalogger/...etc).  The analysis topics are, with our recommendation, unique as they start with the analysis name explicitly so analysis/Economizer_RCx/foo/bar/bim as the topic name versus devices/foo/bar/bin

schandrika, 2020-03-10:
See related issue #995. Querying by source is not a high priority use case for now.

closed 2020-03-10 by schandrika