VOLTTRON/volttron #1166: Should BaseHistorian record source break out objects into sub topics?

kmonson, 2017-01-31:
Currently we take what is published to the record topic and store it verbatim. 

Should we break out objects by key name into sub topics?

This is essentially what we do now with driver publishes to keep things performing well and yet break individual points into their own topics.

The increase in the number of topics is probably not a big deal as we have platforms with a large number of points and it still seems to work well. 

The upside is that it flattens the data and the topic list better reflects the contents of what was published. Flattening out the data by hand and publishing to more topics would bog down the system way too much to get the same structure for the data out of the historian later.

The downside is that we are changing the format of the data in a way that may not be expected and might be considered inconsistent. For instance if a list was published we are in no position to flatten out the contents of that list without context information about the contents of the list.

craig8, 2017-01-31:
Record is meant to be a blob that the user can do with what it is supposed to be.  It requires the end user to know what the dataformat is going in and we make no assumption what so ever...As far as the base historian I thought you said it didn't care about the value.

kmonson, 2017-01-31:
We only care about the type for the devices "all" publish so we can break it out by point. The "all" publish must be a dict.

Yeah let's not do anything with the BaseHistorian then.

closed 2017-01-31 by kmonson