VOLTTRON/volttron #1168: With new router improvements

craig8, 2017-02-01:
Pull request #1097 the following occurs if one updates only one side of a remote platform.

````
2017-01-31 16:30:10,627 (forwarderagent-3.7 7066) forwarder.agent ERROR: Traceback (most recent call last):
  File "/home/volttron/.volttron_10500/agents/c7780360-628e-4087-bfe6-95c01e664bd3/forwarderagent-3.7/forwarder/agent.py", line 263, in publish_to_historian
    message=payload['message']).get()
  File "/home/volttron/volttron_10500/env/local/lib/python2.7/site-packages/gevent/event.py", line 373, in get
    return self.get(block=False)
  File "/home/volttron/volttron_10500/env/local/lib/python2.7/site-packages/gevent/event.py", line 363, in get
    return self._raise_exception()
  File "/home/volttron/volttron_10500/env/local/lib/python2.7/site-packages/gevent/event.py", line 343, in _raise_exception
    reraise(*self.exc_info)
  File "/home/volttron/volttron_10500/env/local/lib/python2.7/site-packages/gevent/_util_py2.py", line 8, in reraise
    raise type, value, tb
Unreachable: VIP error (113): No route to host: pubsub
````

shwethanidd, 2017-12-21:
This issue is fixed. Not reproducible with latest 5.0rc branch. I've tested with forwarder running in 4.0 instance forwarding messages to 5.0rc instance and vice versa. It is working fine for both the cases.

craig8, 2017-12-22:
-

closed 2020-03-10 by schandrika