VOLTTRON/volttron #1184: Anon improvement

craig8, 2017-02-13:
Anonymization data in a historian will require a VolttronCentralPlatform agent to be installed on the instance.

The historian will have a flag for determining whether anonymization should be enabled or not, by default this setting will be False (not anonymizing)

Only the anonymized topics will be written to the database (the others will be cached and an error will be alerted as an error)

schandrika, 2020-03-10:
Is something more than topic replacement string required to achieve the same purpose ?

jhaack, 2020-03-10:
I think this was a check to ensure only anonymized data was written to the DB. Use case is no longer a priority unless someone in the community raises it.

schandrika, 2020-03-10:
No longer a priority. Can be reopened if community has a need for this

closed 2020-03-10 by schandrika