VOLTTRON/volttron #1196: If the key starts with a hyphen ('-')

so3500, 2017-02-17:
Thank you for creating a great open source called volttron.

Http://volttron.readthedocs.io/en/master/core_services/security/Known-Hosts-File.html?highlight=known-host
According to the link, "The known-hosts file should not be edited directly." . However, there was a situation where you had to change the file directly.

`volttron-ctl auth add-known-host --host <host ip: port> --serverkey <serverkey>`
  The command is not executed for instances where you want to perform a command but the <servekey> starts with '-'.

[image]

1. This is the case when problem 1 occurs first.
2. The second was wrapped in "".
3. Third, I changed the initial hyphen to another regular character,

Are you aware of this problem?

craig8, 2017-02-17:
Thank you for posting it this.  I can't believe that we haven't run across this before.  I think if you \- it that would make it work the way it "should" would you let me know if that works?

Thanks

so3500, 2017-11-04:
Would I try again with the key that starts with -?

craig8, 2020-06-16:
#2371