VOLTTRON/volttron #1214: No recovery from database connection failure

acedrew, 2017-02-27:
I had to migrate my mongodb yesterday, and things went a little south, leaving me with a mongo outage of several hours. Once mongo was restored, I was able to see data from existing charts in VC, and assumed all was working. When I looked at it the next morning, all of the data was stale, and on further examination, the VC instance was out of memory. (4GB) Restarting the mongo historian did not resolve the issue and only a full platform restart brought connectivity back, but it of course also lost the data in memory.

craig8, 2017-02-27:
Mongo and vc weren't running on the same system wer they?  4GB of data? logs? ram?

acedrew, 2017-02-27:
4 GB of ram, they are not on the same system, trying to figure out the logs now.

acedrew, 2017-02-27:
OK, working through some things with @craig8 it looks like it did indeed persist the unwritten data to disk, and there wasn't significant data loss, until vc filled it's disks and ram trying to cache it all. Once some disk space was freed up, and the platform restarted, the mongo agent did what it was supposed to and flushed everything to the DB. The problem here is I was unable to find significant messages in the logs saying what happened, and it did not recover on its own without the manual restart.

acedrew, 2017-03-03:
Also believe this is related to #1223

craig8, 2017-03-03:
During testing last night there was also no message when there was a wrong ip address for the mongo database connection @schandrika 

Add connection tests to the main historian framework and add logging for when connection not successfull.

acedrew, 2017-03-04:
That's still not what's causing zmq to run out of file handles and stop attempting to connect, though yes, knowing when the historian is broken would be nice 💃

schandrika, 2019-07-23:
dTested with wrong user, wrong host, and stopping mongo after successful historian start. All these use cases result in multiple error messages in volttron.log.

closed 2019-07-23 by schandrika