VOLTTRON/volttron #1231: crate historian raw

craig8, 2017-03-05:
2017-03-04 19:38:30,400 (crate_historianagent-1.0 29572) volttron.platform.agent.base_historian ERROR: An unhandled exception occured while publishing.
Traceback (most recent call last):
  File "/home/volttron/volttron_22916/volttron/platform/agent/base_historian.py", line 752, in _process_loop
    self.publish_to_historian(to_publish_list)
  File "/home/volttron/.volttron_22916/agents/30d42cda-ba31-45c4-9af5-bcf330fb7864/crate_historianagent-1.0/crate_historian/historian.py", line 329, in publish_to_historian
    insert_data(cursor, topic_id, ts, value, topic_name = topic)
  File "/home/volttron/.volttron_22916/agents/30d42cda-ba31-45c4-9af5-bcf330fb7864/crate_historianagent-1.0/crate_historian/historian.py", line 222, in insert_data
    if self._raw_schema_enabled and table in ("datalogger", "device") and topic_name:
NameError: global name 'table' is not defined

acedrew, 2017-03-05:
I believe this is resolved by #1233 if someone can test on another system than mine.

craig8, 2017-03-19:
OBE for the new schema handles this better.

PR #1233 (2017-03-10): fixed errors checked into develop on crate historian
Resolves #1231 

This change is [[image]](https://reviewable.io/reviews/volttron/volttron/1233)
closed 2017-03-19 by craig8