VOLTTRON/volttron #1234: No support for gz compression in web subsystem

acedrew, 2017-03-05:
Some of our platforms are on wireless WAN connections, it would be nice to have gz so that we were using less bandwidth.

craig8, 2017-05-25:
Links for implementor

https://stackoverflow.com/questions/8506897/how-do-i-gzip-compress-a-string-in-python
https://developer.mozilla.org/en-US/docs/Web/HTTP/Compression
https://en.wikipedia.org/wiki/HTTP_compression

acedrew, 2017-06-03:
Another reason this is important, modern processors are optimized for compression workloads, and it can be faster to compress data before transmitting than transmitting in general, even without hitting network IO. Windows takes advantage of this and actually compresses data in RAM while programs are running.

craig8, 2017-06-04:
I am going to suggest not to take #1379 as is, but instead wait for the discussion on #1378 and epic #1382 to handle this.