VOLTTRON/volttron #1239: VoltTron: holding registers and input registers are not one and the same.

ocschwar, 2017-03-09:
From MasterDriverAgent/master_driver/interfaces/modbus.py:

`    def get_state(self, client):
        #if self.read_only:
        #    response = client.read_input_registers(self.address, count=self.get_register_count(), unit=self.slave_id)
        #else:
        response = client.read_holding_registers(self.address, count=self.get_register_count(), unit=self.slave_id)
`
A lot of pymodbus example code has servers where the same Context object holds the input registers and holding registers, but that is not an inherent part of the Modbus standard. In particular, in SunSpec territory, there are only holding registers, period, although some are marked read only, and some switch their writability on and off. 

We commented this out for our project, but something more general is called for. Perhaps a register attribute that can be set in the registry_configs file ?

kmonson, 2017-04-14:
"A lot of pymodbus example code has servers where the same Context object holds the input registers and holding registers, but that is not an inherent part of the Modbus standard."

That is correct and our code and the pymodbus client do not assume that.

"In particular, in SunSpec territory, there are only holding registers, period, although some are marked read only, and some switch their writability on and off."

I'm under the impression that goes against the MODBUS standard. Registers in the holding register data block should imply read/write capabilities.

If anything I could have called the "Writable" column "Holding/Coil" or something like that and made the Writable column optional and implied its value from the "Holding/Coil" if not provided. 

I have never encountered a device with read-only holding registers and there is no mention of the possibility in the MODBUS specification so I assumed holding implied read-write.

schandrika, 2020-03-10:
@ocschwar does the above reply answer your question? Please feel free to reopen the issue if you need further clarification

closed 2020-03-10 by schandrika