VOLTTRON/volttron #1247: Sometimes points are duplicated during device config

jchap-pnnl, 2017-03-14:
In the VC UI, when configuring devices and scanning for device points, sometimes the points get duplicated when they show up in the table.

We used to prevent that by checking the points and not adding them if they were already in the list. But that produced horrible scalability that led to the UI crashing when we tested it on a device with 6,000 points. So we removed that check, to speed things up, but now you can get duplicate points in the table.

We need to investigate to see what's causing the duplicate points. It doesn't appear to be a bug in the UI code, but that needs to be verified. It might be that the same point is sent across the websocket twice. If so, perhaps we can figure out a way to prevent that.

Another approach for now could be to restore checking if a point is in the list already as a default, but allow an advanced option to turn that check off to improve scalability for devices with lots of points.

acedrew, 2017-03-21:
@jchap-pnnl One way to resolve this on the UI side is build a list that is not directly rendered to the DOM by React, and only flush that into the React app every n seconds or on completion. This will prevent each duplicate check from rerendering the table, and will speed things up considerably.

jchap-pnnl, 2017-03-24:
@acedrew That's actually what we were doing. We wouldn't render any of the points to the DOM until all of the points were received. So we were only checking the points for duplicates in the virtual list in memory. When a new point was received, we would look for it in the entire virtual list up to that point and only add it if the point wasn't already in the list. But even without rendering, that took too long when it was a long list of points. For a list of 6,000 points, at least 18,003,000 comparisons were made.

@craig8 Were we able to verify that removing the check actually sped things up significantly and avoided crashing the browser for the device with 6,000 points?

acedrew, 2017-04-25:
So js stores object properties in an efficient hashmap structure, I would think we could refactor this and use some concatenation of point data to create unique strings that we could check for the existence of, vs searching through a list. If that's what you were already doing, then nevermind 😄 
I can answer for @craig8 as it was my 6000 point device that was slowing down, and yes, getting rid of the duplicate check improved performance substantially.

jchap-pnnl, 2017-05-01:
That's a good suggestion, @acedrew. By storing the points in an object, rather than a list, we could use a unique string for each point as the attribute name for the point item. And we wouldn't have to check if the point was already added, because if the same point was encountered twice, it would just be assigned to the object with the the same name as before, and not create a duplicate item.

A lot of the places where the points list is used in the code, it's iterated through in a loop. To do the same thing with a points object, we could do a loop on the object's keys list: Object.keys(pointsObject).forEach((key) => { doPointStuff(pointsObject[key]) });

But this would be a significant refactor, because the points list is integrated into the code in a lot of places and in multiple ways.

craig8, 2017-05-01:
I suppose the significance is how far you want  DRY code

jchap-pnnl, 2017-05-01:
Hmm, and I wonder if the refactoring might be minimized if we continue to use the points in a list, but use the object only when retrieving them from the scan. That would keep the scanning fast and avoid duplication. And then once the scanning completes, we immediately convert the object to the list using the loop shown above, so we wouldn't have to alter any of the other places the list is used in the code..