VOLTTRON/volttron #1261: Connect local cluster(cloud) with volttron

so3500, 2017-03-27:
Please kindly forward this query (if not appropriate) to the right channel.

I’m working on cloud supported volttron (big data analytic).

Currently I'm working on version 4.0.1.

I would like to 
- Configure local cluster (cloud) and send data of devices topic from volttron to local cluster.
- Then, process collected data via big data processing frameworks (e.g., spark, hadoop) and send messages (i.e, device control) to one of volttron instances.

To do this, I have checked and tried to use cloud agent according to Overview below:
http://volttron.readthedocs.io/en/4.0.1/overview/index.html

However, there seems to be no code implementation. Please kindly let me know any pointer to the codes or (if not exist) kindly confirm which direction I should follow:

1st Approach (use vip)
- install volttron in cloud and use forward agent (or make agent in big data framework interface to connect to volttron)

2nd Approach (use http protocol):
- customize agent (or create a cloud agent our own) to connect it to [local]cloud via http

3rd Approach(use socket): 
- same as 2nd approach but use socket 

Any comments and advice would be appreciable as I await your sincere response.

Thank you in advance!

acedrew, 2020-07-08:
@so3500 Did you get your questions answered? If so, please close.

so3500, 2020-08-25:
@acedrew I solved this problem by implementing a kafka agent.
Thanks!
ref. [How about adding an Apache Kafka agent? #1525](https://github.com/VOLTTRON/volttron/issues/1525)

closed 2020-08-25 by so3500