VOLTTRON/volttron #1292: main__ DEBUG: ValueError: {'type': 'ValueError', 'value': 'missing argument'}

hyper1003, 2017-04-13:
Hi~

I found when I publish a zero value causes "ValueError "
 __main__ DEBUG: ValueError: {'type': 'ValueError', 'value': 'missing argument'

In bellows log, you can see the co2-sensor value is 27.5. 0x41DC0000
I converted it to 4bytes tuples and then made two words  to work around a float value sending difficulty.

When I send a  word  with 0x41DC to point 0 ('>H', csv), then it work well
but tried to send a 0x0000  value to point 1('>H',csv),it causes a Value Error. It happens always.

- logs -
('DB(5) co2-sensor-co2:', 27.5)
DB(6) Hvaci-Temp-setVal-msb: 0x8888
DB(7) Hvaci-Temp-setVal-lsb: 0x7777
['0x41', '0xdc', '0x00', '0x00']
2017-04-13 12:28:04,048 __main__ DEBUG: handle_schedule_request: devices/actuators/schedule/request, {'requesterID': 'platform.actuator', 'priority': 'HIGH', 'min_compatible_version': '3.0', 'taskID': 'platform.actuator-TASK', 'max_compatible_version': u'', 'AgentID': 'platform.actuator', 'type': 'NEW_SCHEDULE'}, [['inu-campus/it-building/gui', '2017-04-13 03:28:04.007595+00:00', '2017-04-13 03:29:04.007634+00:00']]
2017-04-13 12:28:04,049 __main__ DEBUG: Got new schedule request: platform.actuator, platform.actuator-TASK, HIGH, [['inu-campus/it-building/gui', datetime.datetime(2017, 4, 13, 3, 28, 4, 7595, tzinfo=tzutc()), datetime.datetime(2017, 4, 13, 3, 29, 4, 7634, tzinfo=tzutc())]]
2017-04-13 12:28:04,050 actuator.scheduler DEBUG: now_slice in make_current 0
2017-04-13 12:28:04,053 __main__ DEBUG: handle_set: inu-campus/it-building/gui/Co2_msb,platform.actuator, 16860
2017-04-13 12:28:04,053 __main__ DEBUG: _check_lock: inu-campus/it-building/gui, platform.actuator
2017-04-13 12:28:04,069 __main__ DEBUG: ValueError: {'type': 'ValueError', 'value': 'missing argument'}
2017-04-13 12:28:04,078 __main__ DEBUG: handle_set: inu-campus/it-building/gui/Temp_setVal_msb,platform.actuator, 34952
2017-04-13 12:28:04,078 __main__ DEBUG: _check_lock: inu-campus/it-building/gui, platform.actuator
2017-04-13 12:28:04,092 __main__ DEBUG: handle_set: inu-campus/it-building/gui/Temp_setVal_lsb,platform.actuator, 30583
2017-04-13 12:28:04,092 __main__ DEBUG: _check_lock: inu-campus/it-building/gui, platform.actuator
2017-04-13 12:28:09,004 __main__ DEBUG: _update_device_state_and_schedule

- agent.py-
#>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
#       update gui modbus points
#       Co2

        topic=topics.ACTUATOR_SET(campus=self.campus,building=self.building,unit=self.unit,point=self.point_co2_0)
        self.vip.pubsub.publish('pubsub',topic,headers,co2_msb)
#        self.vip.pubsub.publish('pubsub',topic,headers,0x0000)  # co2_msb equals zero makes an missing value error

        topic=topics.ACTUATOR_SET(campus=self.campus,building=self.building,unit=self.unit,point=self.point_co2_1)
        self.vip.pubsub.publish('pubsub',topic,headers,co2_lsb)

#>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

kmonson, 2017-04-13:
I don't think you are using the PUBSUB interface for the actuator correctly. What is the value of co2_lsb?

You should consider the PUBSUB interface deprecated.

I would suggest that you instead use the RPC interface as it is much simpler:

```
point = "campus/building/unit/point"
value = 25.1234
self.vip.rpc.call(
                'platform.actuator',  # Target agent
                'set_point',  # Method to call
                self.core.identity,  # Requestor, ignored in newer versions of VOLTTRON as it's not needed.
                point,  # Point to set
                value,  # New value
            ).get(timeout=30)

```

hyper1003, 2017-04-14:
The value of co2_lsb is 0x0000.

The 0 value makes the ValueError in my test.

Thanks four your suggestion.

kmonson, 2017-04-14:
Is it a string?

hyper1003, 2017-04-14:
It's a interger.

kmonson, 2017-04-20:
Did using the RPC method work for you?

hyper1003, 2017-04-20:
I tried your suggestion using rpc method in actuator.py but when I tested
it it gave me LockError so I am trying to make a new agent. It takes a time
because I am a beginner at this time. I think I can report it next week.

Thanks a lot

kmonson, 2017-04-21:
If you switch from the master branch to the develop branch it will work. Requiring a lock for every interaction was recently removed. The next version of VOLTTRON will see this change.

If you don’t want to switch to develop you can look at the scheduler example agent in the example directory to see a simple example of how to request a lock.

Specifically the RPC call to request a lock is found here: https://github.com/VOLTTRON/volttron/blob/master/examples/SchedulerExample/schedule_example/agent.py#L166

You should also look at the actuator documentation here: http://volttron.readthedocs.io/en/develop/apidocs/services/ActuatorAgent/actuator.html#module-actuator.agent

craig8, 2017-04-24:
@hyper1003 can you please update/close this issue if @kmonson suggestion works out for you?

hyper1003, 2017-04-24:
Sure, I will.

hyper1003, 2017-05-19:
Hi~
We solved this matter!
As your comment, we use rpc.call and it works with zero value.
It took a time to make a our own test agent. thanks for your kind comments. 
I wrote the set_point which I use , it has the rpc call().
I close this issue.

-- ---- set_point() ---
    def set_point(self, topic, value, **kwargs):
        '''_log.debug("Workbench: set_point() is called.")'''
        topic = topic.strip('/')
        path, point_name = topic.rsplit('/', 1)

        try:
            result=self.vip.rpc.call(self.driver_vip_identity,
                                   'set_point',
                                   path, point_name,
                                   value).get()
            return result

        except Exception as e:
            print(e)
            return

closed 2017-05-19 by hyper1003