VOLTTRON/volttron #1321: Make the list of topics historian subscribes to configurable.

schandrika, 2017-04-26:
Currently historian subscribes to devices, analysis, logger, and record topics.  Make this topic list configurable.

kmorri09, 2017-04-28:
I brought a similar issue to the stackoverflow site (https://stackoverflow.com/questions/43621326/volttron-4-1-historian-topic-targeting-and-permissions-comment).

Our system records data from multiple classes of devices (i.e. building temperature/load data and generation meter measurement data), and sends this data to dedicated MySQL databases. So, in addition to a platform historian which is used by our agents, we wish to write data to databases which may be accessed outside VOLTTRON by different groups around UW campus.

It would be useful to have a VOLTTRON historian or database agent that subscribes to specific topics (i.e. devices/BuildingData/* versus devices/SolarData/*) and writes only those topics to the database.

I have implemented my own solution by modifying the code of the historian such that I can pass in "filter_enable" and "topic_list" variables, and the historian only saves to the given MySQL database if enable is True and the topic it hears corresponds to something on the topic_list. I have installed one of these agents per database we are recording data to, each with its own topic_list. I'm happy to provide this code if needed. 

Thanks,
Kevin Morrissey, UW

acedrew, 2017-04-28:
PR PR!

craig8, 2017-04-28:
So @kmorri09 since you have already got the code to do this, can you fork and put that code out and we can take a look and try to get it into the base_historian for all historians?

kmorri09, 2017-04-29:
I did a pull request, adding a SelectiveHistorian agent to examples. Please let me know if you would like to discuss.

schandrika, 2020-03-10:
PR #1964 added custom_topics parameter to base historian config. This can be used to add custom topics that historians can subscribe too.  The recording of default topics - devices, analysis, logger, record - can be turned on or off using the parameters capture_device_data, capture_log_data, capture_analysis_data, and capture_record_data

closed 2020-03-10 by schandrika