VOLTTRON/volttron #1333: Incorporate value scaling as part of driver?

kmorri09, 2017-04-28:
I work with a smart meter, Nexus 1272, in which the modbus data returned needs to be scaled after it is measured. For instance, power is measured as an integer in units of 1/65536 W, I suppose because the manufacturer found it simpler to deal in integers. I am not sure if this is common, but it would be useful for the driver to be able to scale this data so that the values saved to the historian represent the true measurements, instead of a number that needs to be scaled.

I have developed my own solution, in which I include a column in the registry csv file which lists the scaling factor (i.e. 65536). The driver scrapes data to the topic "devices/nexus_meter-unscaled," and I run a separate scaling agent that does the following:

- listens to the topic "devices/nexus_meter-unscaled"
- reads the corresponding scaling factor from the registry csv for each data point
- performs the scaling operation for each point
- republishes the scaled data to "devices/nexus_meter"

I am happy to share my code if it is helpful - I just wanted to point out that this could be a useful component of the driver if this kind of scaling is common in measurement devices.

Thanks,
Kevin Morrissey, UW

schandrika, 2020-03-10:
Is this something common in other measurement devices?

acedrew, 2020-07-06:
Resolved by many scaling options in modbus-tk driver can be closed @jhaack

closed 2020-07-06 by jhaack