VOLTTRON/volttron #1345: Add MonetDB Historian.

ocschwar, 2017-05-05:
MonetDB is a Column oriented SQL database, designed expressly for time series data. I have a working historian agent using it on the back end. 

The speedup it achieves for long topic queries (on the order of 10K datapoints in a server that is tracking 170 topics, well, it's swamped by the data's JSON encoding, but because it stores each topic in its own column, I expect to resolve that, and enforce a type for each column. That would allow a 10x or so speedup. 

I had to modify the insert_stmt in basedb.py and add a monetfuncts.py file. It's in the Fraunhofer fork,

ocschwar, 2017-05-05:
In volttron/volttron/platform/dbutils/, I put modifications to basedb.py and added monetfuncts.py

In volttron/services/core/SQLHistorian/, I added a config.money, 
and modified sqlhistorian/historian.py 

And I added volttron/scripts/historian-scripts/start-historian-monet.sh

The SQL Historian needed to pass the metadata for a topic on to insert_topic so that MonetDB could pick a proper datatype for it. 

So why bother with all this?

Here's why:

MonetDB stores its data by column, not by row. So when you make a query for a particular topic, it gets the topic very quickly, even for long-ranging queries. 

Since it stores everything in its own column, you can use SQL types rather than JSON encoding for each data point, which speeds things as well. 

MonetDB is able to add and remove columns on the fly without a performance hit, so new topics are not a problem. 

All this is available here:

https://github.com/FraunhoferCSE/volttron

My Volttron instance is accumulating data right now. I'll have a benchmark for comparison tomorrow.

acedrew, 2017-06-04:
@ocschwar this looks really exciting, any updates on performance?

ocschwar, 2017-06-07:
As I'm no longer on this project, this will have to be my closing notes: 

The bad: 

1. MonetDB's queries are considerably faster than MySQL or SQLite, but I've learned hte hard way that they are not the bottleneck in getting data out of the Historian. 

2. To use a column based historian, you have to add new columns for each new topic, so this only works on SQL databases where "alter table data add column new_topic" can work quickly. THis works very nicely with MonetDB, but is not the end of the story. 

2. You need a Historian that will take incoming messages from the topic bus and store in bulk, not one point at a time. I did not get far enough to write that. What this means is that if the  master driver agent is posting breadth-first-all, the Historian needs to do "insert into data (ts, topic1, topic2, topic3) values (timestamp, datapoint1, datapoint2, datapoint3)"

To get MOnetDB historian to match the same pattern as the others, what I did was for each single topic, try an insert of a row with timestamp and one datapoint, and if that failed, do an update. That was just not performant, and it didn't help that MonetDB does not have an "upsert" syntax yet.

I still think this shows a lot of promise. Queries are lightning fast, and whole row inserts would fix the performance problem I was seeing. And, MonetDB is aggressive about compressing columns of unchanging data, which for a MOdbus master driver agent is a very good thing. Data accumulation was slow. FInally, for removing old data points and freeing hard drive space, it shines.

acedrew, 2017-06-07:
Thanks for the feedback, I appreciate the additional information. I think most of the benefits you saw are being offered by the CrateHistorian we've been working on with PNNL, but it's nice to know there's another promising solution if Crate doesn't work out.

schandrika, 2020-03-10:
Now that we have CrateHistorian working do we still need to consider this? 
I feel this can be revisited if there is more interest in column oriented database

jhaack, 2020-03-10:
I agree. Close for now unless it gets raised again

closed 2020-03-10 by schandrika