VOLTTRON/volttron #1374: WeatherAgent Conversation

jhaack, 2017-05-30:
Several efforts are creating their own Weather Agent. This issue is meant to track the different code bases and use cases.

jhaack, 2017-05-30:
PNNL Developed alternative:

High level description:
WeatherAgent2 uses publish/subscribe to provide the data. Consumer requests weather data by publishing to a certain topic and listen to another topic for reply. This agent is to leverage free weather data sources (not paid ones).

Features:
-	Provides historical, current, forecast data
-	Provides data from multiple sources (Wunderground, Noaa, Tmy3, etc.)
-	Provides variety of data temperature, wind, solar irradiation, etc.

Request params:
-	Wunderground key
-	State, City, Zipcode, etc.
-	Interested points or All points
-	Data source
-	Time frame 

Why we need data from multiple sources? Use cases that one source can’t satisfy:
-	Need temperature forecast (Wunderground) and solar irradiance forecast (Tmy3)
-	Historical temp, windspeed (Wunderground) and historical solar data (NOAA)
-	Current weather and typical year weather (Tmy3) data for weather normalization
-	Etc.

craig8, 2017-06-04:
From @kmorri09  https://github.com/VOLTTRON/volttron/pull/1368

> Our agent uses the weather underground base agent to get current weather every 5 minutes, as well as forecasts for temperature, sky cover, and wind speed every hour (24-hour forecast). The weather data is written to a csv file for a given month (4 csv files per month, one for conditions and 3 for forecasts). At the start of a new month, new csvs are created.
> 
> Included are example output files.

craig8, 2017-06-04:
From @maskedpatel https://github.com/VOLTTRON/volttron/pull/1362 with readme located https://github.com/maskedpatel/volttron/blob/91cdf6677cc53644cf8139ec1f07dc05f1c46168/services/core/weatheragent/README

shwethanidd, 2020-02-20:
Weather underground agent is deprecated since Weather Underground service is no loner a free service

closed 2020-02-20 by shwethanidd