VOLTTRON/volttron #1395: uJSON & Forwarder Agent

thibeco, 2017-06-09:
Here is a stacktrace of an issue with data being pulled from a BACnet network and forwarded to a VC instance.

This is on Volttron 4.1 (`master` branch).

```(bacnet_proxyagent-0.2 18251) bacnet_proxy.agent ERROR: ERROR DURRING SCRAPE (Class: object Code: unknownObject)
(bacnet_proxyagent-0.2 18251) <stderr> ERROR: ERROR:bacnet_proxy.agent:ERROR DURRING SCRAPE (Class: object Code: unknownObject)
(forwarderagent-3.7 18250) <stderr> ERROR: Exception in thread Thread-2:
(forwarderagent-3.7 18250) <stderr> ERROR: Traceback (most recent call last):
(forwarderagent-3.7 18250) <stderr> ERROR:   File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
(forwarderagent-3.7 18250) <stderr> ERROR:     self.run()
(forwarderagent-3.7 18250) <stderr> ERROR:   File "/usr/lib/python2.7/threading.py", line 754, in run
(forwarderagent-3.7 18250) <stderr> ERROR:     self.__target(*self.__args, **self.__kwargs)
(forwarderagent-3.7 18250) <stderr> ERROR:   File "/<path_to_volttron>/volttron/platform/agent/base_historian.py", line 740, in _process_loop
(forwarderagent-3.7 18250) <stderr> ERROR:     backupdb.backup_new_data(new_to_publish)
(forwarderagent-3.7 18250) <stderr> ERROR:   File "/<path_to_volttron>/volttron/platform/agent/base_historian.py", line 1004, in backup_new_data
(forwarderagent-3.7 18250) <stderr> ERROR:     (timestamp, source, topic_id, dumps(value), dumps(headers)))
(forwarderagent-3.7 18250) <stderr> ERROR:   File "/<path_to_volttron>/volttron/platform/agent/base_historian.py", line 260, in dumps
(forwarderagent-3.7 18250) <stderr> ERROR:     return ujson.dumps(data, double_precision=15)
(forwarderagent-3.7 18250) <stderr> ERROR: OverflowError: Invalid Nan value when encoding double
(forwarderagent-3.7 18250) <stderr> ERROR: ```

craig8, 2017-06-09:
Note removing ujson via pip remove ujson was able to fix the issue.  This will be fixed shortly.

craig8, 2017-06-19:
@thibeco the fix to this was merged in issue #1399 into the develop branch.  Please update your instance.

closed 2017-06-19 by craig8

thibeco, 2017-06-22:
@craig8 I updated an instance and it was able to successfully forward data.

craig8, 2017-06-22:
@thibeco awesome thanks for the feedback!