VOLTTRON/volttron #1419: Historian returns wrong result format for multi topic query when only one of the topic has valid results

schandrika, 2017-07-14:
When querying historian with multiple topics and only one of the topic returns valid results, historian returns the matching records/rows without the topic name information so the user won't know which topic the records belong to. 

**Expected result for single topic query:**
{ 'values':   [[timestamp, value], [timestamp,value],....],
'metadata':{key:value,...}}
**Expected result for multi topic query**
{ 'values':   {'topic_name': [[timestamp,value], [timestamp,value],...], 
   'topic_name':[}
'metadata':{key:value,...}}

**Current out put for the error scenario**
{ 'values':   [[timestamp, value], [timestamp,value],....],
'metadata':{key:value,...}}

fstshrk, 2017-07-14:
Good catch!

-----------------------------------------------------------------------------------------------------------------------------
   __o       Bora Akyol, Ph.D. Pacific Northwest National Laboratory
 _`\<,_
(_)/ (_)     bora@pnnl.gov<mailto:bora@pnnl.gov>, +15093716682

craig8, 2017-07-15:
Which specific historian are you working with?

hlngo, 2017-07-15:
I am seeing the issue with SQLite but I was querying through Volttron query interface so I guess that happens at query interface level. I am not 100% sure though. Chandrika and I talked about it yesterday and she may have some more info since then.

schandrika, 2017-07-17:
Fails for both SQL and Mongo historian.  Should I be testing crate historian too. Currently non of the generic historian  test cases are being run for cratehistorian.

closed 2017-08-25 by craig8