VOLTTRON/volttron #1428: Add "edit" subcommand to vctl config command.

kmonson, 2017-08-03:
1. Gets raw file from store and saves it to a temporary file.
2. Open raw file in editor. (How do we pick the editor?)
3. Wait until editor quits.
4. Attempt to write file back to store.

fstshrk, 2017-08-03:
Kyle

This needs to handle file locking so that we don’t have two different terminal sessions editing the file at the same time.

-----------------------------------------------------------------------------------------------------------------------------
   __o       Bora Akyol, Ph.D. Pacific Northwest National Laboratory
 _`\<,_
(_)/ (_)     bora@pnnl.gov<mailto:bora@pnnl.gov>, +15093716682

kmonson, 2017-08-03:
Hmm, currently there is no way to "lock" a file in the store. It would be a last in wins sort of situation. I'm not super excited about creating a new feature in the store that could leave an agent's store in a bad state. If the edit command goes wrong and it fails to release it's lock what do you do then?

There is a per agent store lock, but it is designed to only be held for less than a second (10 seconds at the worst) and is not meant as a long term lock. It blocks out all other config changes for that store until the agent has had a chance to process the most recent change to the store.

kmonson, 2017-08-03:
I could certainly check if the temp file was actually modified before committing changes.

fstshrk, 2017-08-03:
That is a workable solution for now.

-----------------------------------------------------------------------------------------------------------------------------
   __o       Bora Akyol, Ph.D. Pacific Northwest National Laboratory
 _`\<,_
(_)/ (_)     bora@pnnl.gov<mailto:bora@pnnl.gov>, +15093716682

commit 32c9c527 (2017-08-08): Add edit subcommand to vctl config. Fixes #1428
commit 75d8a356 (2017-08-08): Merge pull request #1431 from kmonson/feature/config-edit

Add edit subcommand to vctl config. Fixes #1428
closed 2017-08-24 by kmonson