VOLTTRON/volttron #1444: Upgrade .travis.yml file

craig8, 2017-08-15:
End of life for precise so we should update the .travis.yml file to only build for trusty.  We should include higher versions of python e.g. 2.7.13 to see what effect that has.

In addition we need to add the databases to the build so that we test them on ci.

bonicim, 2020-05-26:
Since the last comment on this issue, Travis now uses Xenial as the default. Since [Python 2 is no longer supported](https://www.python.org/doc/sunset-python-2/), I recommend that the travis.yml file be configured only with 3.x and above. I will submit a PR to reflect this.

craig8, 2020-05-26:
See if you can use tests from 3.6-3.8 in the matrix.  Also, bionic should be tested whether through a docker based test image or through the travis-ci capability.  I haven't looked to see if the free tier allows bionic in non-dockerized builds

bonicim, 2020-05-26:
Tracking on bionic. Will look into it.