VOLTTRON/volttron #1448: Volttron should report invalid configuration

craig8, 2017-08-18:
VOLTTRON should log the configuration when starting up at info level.  If any property in the config file is invalid it should report at ERROR level and then stop the platform.

acedrew, 2017-08-18:
This is the same issue as #1322

mikeroup, 2017-08-30:
I have logging handled but we can't really take care of the regular config file since it gets read in parse_args, before logging is set up.

craig8, 2017-08-31:
Print to standard error if there is an incorrect value in the config file i.e. an unknown key 

If someone puts in

vpi-address = tcp://blah:22916

Then the platform should not blissfully start

craig8, 2017-08-31:
@acedrew slightly different but definitely related as the configuration versus the logging config is two different files...but @mikeroup  is going to make everything good with this anyways!

craig8, 2017-09-14:
@acedrew how do you determine if the logging file is invalid?

schandrika, 2020-03-10:
Fixed in PR #1472

closed 2020-03-10 by schandrika