VOLTTRON/volttron #1449: README.md "Next Steps" links broken

kuhrusty, 2017-08-18:
Hey, when looking at https://github.com/VOLTTRON/volttron, clicking on the documentation links under "Next Steps" gives "Permission Denied" pages from volttron.readthedocs.io.

(Apologies if this is the wrong place for this!  I didn't see an existing bug filed for this.  It looks like those lines of README.md were last modified 10 months ago, so presumably the problem is some change on voltron.readthedocs.io.)

craig8, 2017-08-19:
You are right they are.  We have fixed them in our development branch but that hasn't been vetted out to master yet.  However I think that I will go ahead and take care of that sooner rather than later.  For now you can use http://volttron.readthedocs.io/en/latest/ and go from there as that would be the main site pages.

craig8, 2017-08-19:
will be taken care of in #1450

closed 2017-08-23 by craig8