VOLTTRON/volttron #1469: Add visibility to BaseHistorian backupDB size for monitoring

acedrew, 2017-09-08:
In situations where the ingestion rate of the historian DB is slowed, the backupdb acts as a queue for records to be persisted, currently there's no insight into the depth of that queue.

craig8, 2017-09-08:
Its like you have read my mind on this.  I am thinking we don't want to incur the penalty of getting exact numbers of records in the queue, but instead just report the size of the cache file?  This could be sent on the message bus and/or logged to the log (as info) on perhaps an hour/half hour basis? 

Should the logging go to a historian log?  If backed up that seems like it wouldn't get through so not sure that putting it in one of the historian topics would be good.  But I do feel it should go to the message bus.  I am thinking a new topic say ops or metrics or something that would be known to be reported on by these types of things.

acedrew, 2017-10-02:
Yeah, that makes sense, I'd be happy to work on at least one or two agents that could send out-of-band from the metrics topic to various aggregation platforms (Influx, Prometheus, Graphite)

schandrika, 2020-03-10:
Currently there is a alert sent out when cache size 90% full.  The size of backup db and how when to report that the cache is getting full can be configured using the parameters backup_storage_limit_gb and backup_storage_report. By default there is no backup_storage_limit_gb, if the backup storage limit is set backup_storage_report is 0.9 by default. 

We also report the cache count as part as health status in the process loop. So the cache count is updated periodically as part of health status. Agent can do a get_status or register status change call back. Would these together sufficient for this use case?

craig8, 2020-03-10:
I am thinking that this is before the health subsystem was extended to the base historian.  So I would say this is resolved.

closed 2020-03-10 by schandrika