VOLTTRON/volttron #1473: Installation error

srisankethu, 2017-09-10:
Tried installing volttron following the instructions given [here](https://volttron.readthedocs.io/en/latest/install.html) but facing an error. I am using Ubuntu 17.04 OS. The error seems to related to proxy. I have set up env variables ``http_proxy``, ``https_proxy``. But still facing the same error. Can someone please help me out?

kmonson, 2017-09-11:
Are you seeing an error message?

srisankethu, 2017-09-12:
On running ``python2.7 bootstrap.py --force`` -> https://pastebin.com/hBhk6YHS

craig8, 2017-09-12:
Thanks for the pastebin.  That will really help.  In addition, I am having one of our students build us a vm using ubuntu 17.04 just like what you are using and we will hopefully see the same issue and then can take corrective action.

I believe since you have already got an env directory you can activate the environment and then run the python bootstrap.py.  This will upgrade the volttron instance without removing the python environment.

srisankethu, 2017-09-13:
Thank you @craig8 :+1:  ``export all_proxy="https://<proxy>:<port>/"`` worked

craig8, 2017-09-13:
Awesome....glad that worked for you.

closed 2017-09-13 by craig8