VOLTTRON/volttron #1483: Foward Historian background thread error

craig8, 2017-09-19:
Steps to reproduce

  1. Create a forwarder from one instance to a second instance.
  1. Allow master driver to publish on the first instance such that the second instance receives data.
  1. Stop the second instance.

The following trace will occur, and the second thread of the forward historian/base historian will have exited and will not resume until restarting agent (possibly reconfiguration could restart it).

````
2017-09-18 22:35:40,032 (forwarderagent-4.0 3061) <stderr> ERROR: Traceback (most recent call last):
2017-09-18 22:35:40,033 (forwarderagent-4.0 3061) <stderr> ERROR:   File "/home/volttron/git/volttron/env/local/lib/python2.7/site-packages/gevent/greenlet.py", line 618, in _kill
2017-09-18 22:35:40,033 (forwarderagent-4.0 3061) <stderr> ERROR:     waiter.switch()
2017-09-18 22:35:40,034 (forwarderagent-4.0 3061) <stderr> ERROR:   File "/home/volttron/git/volttron/env/local/lib/python2.7/site-packages/gevent/hub.py", line 843, in switch
2017-09-18 22:35:40,035 (forwarderagent-4.0 3061) <stderr> ERROR:     assert getcurrent() is self.hub, "Can only use Waiter.switch method from the Hub greenlet"
2017-09-18 22:35:40,036 (forwarderagent-4.0 3061) <stderr> ERROR: AssertionError: Can only use Waiter.switch method from the Hub greenlet
2017-09-18 22:35:40,037 (forwarderagent-4.0 3061) <stderr> ERROR: <callback at 0x7fe7942d0c10 args=(<Greenlet at 0x7fe795b53c30>, <class 'greenlet.GreenletExit'>, <gevent.hub.Waiter object at 0x7fe7942d1b40>)> failed with AssertionError
2017-09-18 22:35:40,038 (forwarderagent-4.0 3061) <stderr> ERROR: 
````

craig8, 2017-09-19:
Note this issue should be debugged from https://github.com/craig8/volttron/tree/forwarder-pause as there are several bug fixes that were fixed in this version (it's based from develop)

mikeroup, 2017-09-19:
Trace occurs in the datamover as well when the second instance is shut down.

craig8, 2017-09-22:
https://github.com/kmonson/volttron/commit/d02fd7f22db7a0128cd9870d717d72da21535776 does fix this issue...but does not fix #1485

kmonson, 2017-09-25:
Hey, I said "probably". I didn't say it would for sure.

commit d02fd7f2 (2017-09-27): Fixes #1483. Probably fixes #1484 and #1485
closed 2017-10-18 by craig8