VOLTTRON/volttron #1484: Spurious error when using config edit to create a new config

jhaack, 2017-09-19:
RPC error is thrown, but new config entry is created and can then be edited. Error just being thrown on a check and could cause confusion.

```` 
('RPC ERROR', 'Traceback (most recent call last):\n  File "/home/jer/git/volttron-craig/volttron/platform/vip/agent/subsystems/rpc.py", line 169, in method\n    return method(*args, **kwargs)\n  File "/home/jer/git/volttron-craig/volttron/platform/store.py", line 256, in manage_get_metadata\n    raise KeyError(\'No configuration file "{}" for VIP IDENTIY {}\'.format(config_name, identity))\nKeyError: \'No configuration file "config" for VIP IDENTIY forwarder2\'\n')
`````

kmonson, 2017-09-19:
This is a result of trying to get the meta data for a file that does not exist. I'm not sure how to fix this other than returning error codes instead of raising an exception.

kmonson, 2017-09-19:
I guess we could always return None if it doesn't exist.

PR #1494 (2017-09-27): Added config store to historians
Fixes #1354 - pause forwarder
Fixes #1484, #1485, #1400 83

Added methods for creating and validating publishes of device messages for tests.
<a href='#crh-start'></a><a href='#crh-data-%7B%22processed%22%3A%20%5B%22https%3A//github.com/VOLTTRON/volttron/pull/1494%23discussion_r141445871%22%2C%20%22https%3A//github.com/VOLTTRON/volttron/pull/1494%23discussion_r141446729%22%2C%20%22https%3A//github.com/VOLTTRON/volttron/pull/1494%23discussion_r141447610%22%5D%2C%20%22comments%22%3A%20%7B%22Pull%20efb914fd3040ccb413050e76b719144393e145fb%20services/core/ForwardHi [... cut]
commit d02fd7f2 (2017-09-27): Fixes #1483. Probably fixes #1484 and #1485
closed 2018-03-30 by craig8