VOLTTRON/volttron #1491: Rebuild in place script

jhaack, 2017-09-26:
Would it make sense to have (or do we already) a script that updates code for an agent only without requiring you to add a config file.

Use case is: I want to rebuild my 4.1 agents for 5.0 but don't want to redo config files or keys.  Of course, this would fail if config file requirements have changed.

acedrew, 2017-09-26:
The way I handle this is my deployment leaves config templates in the
/var/lib/volttron directory and I generate a script for each agent that
will reinstall using that config file. Not saying this is a solution, but
upvoting the need/offering a demonstration paradigm.
Andrew

craig8, 2017-09-26:
install-agent doesn't require a config file, but it also doesn't reuse a config file from a currently installed agent.  I think that Andrew's way is the same way we do it, except we just have ours in a config directory.  I think it is good practice to separate the config from the agent tree, however there should be better documentation on how users can automate install/reinstalling of agents.

craig8, 2017-09-26:
Also it is very easy to build an agent that just sits and waits for configuration before it actually does something and that might be a good paradigm to go for.