VOLTTRON/volttron #1513: Time incorrect on Volttron Central graphs

dmossie, 2017-10-10:
Our volttron central graphs started saying "Oct 2" (8 days ago?) when we hover over the recent points. However the time on our Volttron Central and our collecting instances know it is Oct 10. The data still seems to be saved in the database correctly though.

[image]

That is the most recent data, I found it strange because it says updated a few seconds ago, and it does update with the right time but not the date. Our collecting instance is on develop, and our central is on master which could be the issue.

This issue just started sometime recently and I do not believe has been a long term problem

craig8, 2017-10-18:
@dmossie I believe we figured out that your forwarder had gone offline correct?  If so lets close this.

dmossie, 2017-10-18:
Hey @craig8, yes my forwarder had gone offline, however the issue is that the new data coming in have these old timestamps as well.

However it only says the wrong timestamps in Volttron Central. In the database the times are correct.

The new dates coming in got reset to Oct 1 on Monday some how. So the new data today are saying Oct 3. I just restarted the data collector and the central and it still says Oct 3 so I am not sure how they got changed but I'll keep an eye on it...

dmossie, 2017-10-19:
Also both our Volttron Centrals on different servers have this issue.