VOLTTRON/volttron #1525: How about adding an Apache Kafka agent?

so3500, 2017-10-14:
kafka is a distributed platform capable of handling large volumes of messages.

I think these platforms will be useful for VOLTTRON, which needs to collect and process large amounts of data.

Currently I'm testing the example of agent.

This agent has the following functions.

1. Send subscribed data from MessageBus to apache kafka broker

2. Publish the data received from the kafka broker to MessageBus

The link below is the official site of apache kafka.
https://kafka.apache.org/

How about adding an Apache Kafka agent?

so3500, 2017-10-14:
Apache Kafka supports the JAVA API. However, I implemented one using one of the famous libraries that support Python.
https://github.com/dpkp/kafka-python

craig8, 2017-10-16:
Sounds like you have the experience to build that agent.  It would really cool to integrate with more open source projects!

so3500, 2017-10-19:
Thank you very much for showing positive response.

I'll send a pull request after an additional simple test and source code modification.

Could you please give me the necessary precautions for a pull request?

Thank you.

jhaack, 2017-10-19:
Are you on the VOLTTRON office hours? I think this would be a good contribution to highlight next meeting.

so3500, 2017-10-20:
Are you referring to this link?
https://volttron.org/office-hours

Sadly, there is something I have to deal with at the time.

craig8, 2017-10-23:
http://volttron.readthedocs.io/en/releases-4.1/community_resources/contributing.html#contributing is our contributing documentation...

Make sure you are working against the develop branch.  In general your code should go into a contrib folder such as https://github.com/VOLTTRON/volttron/tree/develop/services/contrib.  There are two user contributed agents in there already.  

There is a "general" agent builder that can be accessed via volttron-pkg init.  From that you will see the newest method for creating from an agent template.

so3500, 2017-10-24:
Thank you for the detailed instructions.
I'll update the description and code for the kafka Agent by referring to the contributing documentation.

so3500, 2017-11-07:
I am currently testing an Kafka agent and have an unknown error.
Below is a picture of Kafka agent.
[image]

And the picture below is the big picture that I think (Kafka agent will be the picture like below if you think the data collected from Voltron is collected from the cloud or center!
[image]

Anyway, the function is good. I have confirmed that both of the above functions work well. 

1. Send subscribed data from MessageBus to apache kafka broker and then, comsumer is received from broker
[image]

2. Publish the data received from the kafka broker to MessageBus and then, check with the listener agent.
[image]

But, There is a json error log in the log file. When I contacted kafka-python, have a look at the logging system of the Voltron internal system. 
https://github.com/dpkp/kafka-python/issues/1228

[image]
[image]

Could you help me? Do I need to pull the code once?

so3500, 2017-11-08:
Thank you for your reply.

Well, there is one more issue. 

Currently, the consumer of KafkAgent checks for the existence of a message before fetching it from the broker. However, I used the periodic annotation provided by VOLTTRON to perform this logic (whereas in VOLTTRON, each agent seems to perform the asynchronous method through the callback function when communicating with the message bus) The most ideal way seems to be using asynchronous communication.

According to my research, the asynchronous libraries available in the python2 version are [gevent](http://www.gevent.org/) , [Tornado](http://www.tornadoweb.org/en/stable/).

Is there a good idea to allow VOLTTRON to communicate asynchronously with external systems ??

kmonson, 2017-11-08:
VOLTTRON is built on top of gevent so you are free to use that in your code. You could start a greenlet that runs a processing loop that blocks until something comes in from the broker. That is basically what we do with ZMQ right now for the message bus.

sonjinho, 2018-04-10:
Ummm The error should be check this url  (https://github.com/VOLTTRON/volttron/blob/master/volttron/platform/agent/utils.py#L372)

In my Opinion VOLTTRON logging Format Setting is JsonFormat .

schandrika, 2020-03-11:
Kafka Agent is now part of volttron core. It is under volttron/core/contrib

closed 2020-03-11 by schandrika