VOLTTRON/volttron #1544: Issue on the installation of bacnet proxy agent (solution found)

geyin96, 2017-10-24:
When I packed up the bacnet proxy agent in volttron 4.1, the error shows:
[image]

Then I fixed it by modifying the code in async.py as below:
[image]

kmonson, 2017-10-24:
This happens when the platform Python is compiled in such a way as to remove all docstrings from the standard library. We were aware of the problem and I could _swear_ we had fixed it. We'll get the fix into 5.0 rc branch shortly.

craig8, 2017-10-24:
@geyin96 you mention you had this issue in volttron 4.1 would you be willing to create a pull request with this fix in it to patch the version you are using...then @kmonson can propagate that to the other versions.

closed 2022-04-29 by bonicim