VOLTTRON/volttron #1550: How about creating separate log file for each agent?

YingyingTang111, 2017-10-27:
Currently we can use volttron.log to check the printouts/logs from all agents.
It will be helpful if we can have each agent outputs separately its log file.

craig8, 2017-11-01:
Do in this case is to have multiple terminals open with the following
````
tail -f volttron.log | grep listener
````
````
tail -f volttron.log | grep otheragent
````

Another option would be to use a log aggregation framework such a  as elk to query the different agent's logs.

Finally you can do that by setting up a different logging configuration file whereby each agent would have a different filehandler.  Note we haven't tried this but in theory it should work out.  I may do some research and see if I can come up with something.

so3500, 2017-11-19:
One way is to have the logs start with the name of each agent.
```
_log.info ('Actuator_agent: {0}'. format (<log>))
```
```
cat volttron.log | grep Actuator_agent
```

In my case, I was able to verify that the function worked well by having the log start with the agent's function name.
```
_log.info ('My_function: {0}'. format (<log>))
```
```
cat volttron.log | grep My_function
```

schandrika, 2020-03-11:
There are several ways this can be achieved without code changes. Closing issue

closed 2020-03-11 by schandrika